ottowayi/pycomm3 #15: Asking about slc.py update

SagarPawar606, 2020-02-26:
Hey ottowayi and all other contributors, first of all, I appreciate all of your work here. I am currently working on a project with Rockwell Micrologix 1400 plc with python 3 environment. And I needed slc.py to work with this project, so I am asking you all that is there any plan to update that, its very urgent and important for me. As I'm not that great at python else I would have done that myself and contributed to your work. 
Thanks in advance !

ottowayi, 2020-02-26:
I don't have any plans at the moment to work on it, just due to time and other work I have.  My company doesn't use MicroLogix and we only have 1 SLC, so I wouldn't be able to test it very well.  I haven't removed any of the code, but changes elsewhere may have broken it.  If you want to try getting it working, it might be fairly easy.  I will take a quick look and fix some import errors and stuff and make a branch for you to test from, but I can't make any promises it will work.

ottowayi, 2020-02-26:
So I made a branch called 'slc'.  I did a quick run thru and fixed any errors and did a quick test.  It looks like the read_tag method works, but not write_tag.  Unfortunately I don't have anytime to look into it, but you can try it out if you want.  Else, you can look into using pylogix, I believe they are actively support MicroLogix.

```    
with SLCDriver('10.61.50.11') as plc:
        print(plc.read_tag('N7:1', n=5))
        plc.write_tag('N10:10', 1)
```

SagarPawar606, 2020-02-26:
Thanks for your time and effort, but I don't think that pylogix will work, as they clearly mentioned "This project does not support PLC5, SLC or the Micrologix controllers." in their doc.

ottowayi, 2020-02-26:
Oh you're right, just the MicroLogix 800.  Check [this](https://github.com/bpaterni/pycomm/tree/pycomm3) fork of pycomm, it's the original Python 3 translation I forked this repo from.  I believe they mostly did work on the SLC driver, so it may work or hint at what I broke.

ottowayi, 2020-03-02:
I'm leaving this issue open in case anyone wants to work on it.

AndrewStuhr, 2020-06-10:
Hello, do you have any expectation when MicroLogix support will be ready?  Thanks!

ottowayi, 2020-08-07:
see the latest release (0.10.0) for a working (hopefully) SLCDriver.  I did some initial testing with a SLC, but I don't have any MicroLogix hardware to test with.  Feel free to try it out and let me know if there's issues with it.

closed 2020-08-07 by ottowayi