ottowayi/pycomm3 #21: Looking for virtual PLC

gilzin, 2020-03-18:
First, great work by the pycomm developers! 
Can someone point me to a virtual PLC that I can simulate/send commands with pycomm3 ?
TNX

ottowayi, 2020-03-18:
Unfortunately, you cannot use Studio Emulate (or whatever they call it now). It doesn't support Ethernet/IP comms at all.  One option would be SoftLogix, I haven't tried it personally, but it should work in theory.

gilzin, 2020-03-18:
Thank's for the quick reply!
I am looking for some Linux distro to run a virtual cip server and I think that Soflogix only have Windows one

ottowayi, 2020-03-18:
Oh yeah that might be difficult, this library is really only for Rockwell PLCs and Rockwell is Windows only.  The library will run on linux, but Rockwell's stuff will not.  Have you looked into CPPPO?  I think you can run a simulator server with it, but I have no idea if this library will work with it or not. I looked at it a long time ago, but then I found pycomm and decided to use that instead.  If you do find that it works, I would be more than happy to include an example for others.

gilzin, 2020-03-18:
LOL....I was just looking and installing CPPPO :-) 
It kind of communicates, CPPPO working as a server but when I am asking for example for the PLC info or trying to read some tags, I can see that the server is reacting but doesn't giving what I am expecting to see.
[image]
server-side

[image]

client-side

my code in the client-side
[image]

Any thoughts?

ottowayi, 2020-03-18:
It looks like the forward open is failing, you can try using ``large_packets=False`` in the constructor.  Also, could try also ``init_info=False`` and also ``plc.use_symbol_ids = False`` too.  I'm not sure what features it supports or not.

gilzin, 2020-03-18:
Thank you , I will try it

closed 2020-03-20 by ottowayi