ottowayi/pycomm3 #24: Is there a way to find all instances of a certain udt?

Dreistul, 2020-03-24:
I see how I can look at the whole tag list, and I see how an individual tag will say its tag type , but what would be the best way of finding all tags using a certain UDT?

ottowayi, 2020-03-24:
Yeah, you would want to do something like this:

    udt_tags = ( tag['tag_name'] for tag in plc.tags
                        if tag['tag_type'] == 'struct' and
                           tag['data_type]['name'] == 'TheTypeImLookingFor')

I'm working on creating some good documentation with examples for different use cases, but it's still a work in progress.  Hopefully soon I'll have it on Read The Docs with plenty of information.

Dreistul, 2020-03-25:
after adding the missing single-quote of the third line,  I received the following when I try "print(udt_tags)".    I'm not quite python savvy enough to know what this means:
<generator object <genexpr> at 0x0000026E36B527B0>

ottowayi, 2020-03-25:
that's because it's a  [generator](https://wiki.python.org/moin/Generators), if you do ``list(udt_tags)`` or replace the outer parentheses with brackets to make a list.

My example uses a [comprehension](https://python-3-patterns-idioms-test.readthedocs.io/en/latest/Comprehensions.html), it's a simpler/faster way to make a list rather than using a for loop.  It's equivalent to doing:

    udt_tags = []
    for tag, _def in plc.tags.items():
        if _def['tag_type'] == 'struct' and _def['data_type']['name'] == 'TheTypeImLookingFor':
            udt_tags.append(tag)

Dreistul, 2020-03-25:
when I tried list(udt_tags), I got the following: 
Unhandled Client Error
Traceback (most recent call last):
  File "<ipython-input-31-b11bd9fcc4a6>", line 14, in <module>
    seq_list = list(udt_tags)
  File "<ipython-input-31-b11bd9fcc4a6>", line 11, in <genexpr>
    udt_tags = ( tag['tag_name'] for tag, _def in plc1.tags.items()
TypeError: string indices must be integers

The for-loop approach worked for me, though.

ottowayi, 2020-03-25:
oh sorry, yeah my example is incorrect.  It should just have ``tag``, not ``tag['tag_name']``.

closed 2020-03-27 by ottowayi