ottowayi/pycomm3 #31: forward_open failed, UDT names without values, and multi-reading

AndrewStuhr, 2020-04-20:
Sweet library!

I have 3 questions:

1.) Do you know why I'm receiving this message- "forward_open failed - Service not supported - Extended status info not present".  The code is working despite this message.  Here is my script and output:

`
from pycomm3 import LogixDriver

ip 			= '192.168.50.29'
tagName 	= 'DAQ_Batch_Number'
udt			= 'DAQ_UDT'

with LogixDriver(ip) as plc:
    print(plc.read(tagName))
    print(plc.read(udt))

`

[image]

2.) Is it possible to only read the names/type of UDT children, but not their values?  This may seem odd, but I'm just concerned about performance.  If a UDT is huge, that could be a lot of data comin back if we just need children names.  Any thoughts?

3.) Is there a limit to how many tags can be read when calling plc.read(*tags)?  I believe PyLogix does have a limit for multi-reads.

I'm really excited about this library!  I didn't know it existed until your recent post.

Thanks

ottowayi, 2020-04-20:
1 - You're probably connecting to a PLC that doesn't support the Extended Forward Open (4KB connection) and it's failing back to the standard one which is only 500 bytes.  If you're using an ENBT or firmware below v20 it will happen.  It's not really an error, so you can safely ignore it or you can add the ``large_packets=False`` option when making the LogixDriver to prevent trying the larger connection first.

2 - Yeah, when it opens the connection it actually reads all of the controller scoped tags and their definitions.  This allows the read/write methods to only require the tag names.  If you just want the udt attributes, you can do ``plc.data_types[UDT]``.  The tag list is available from ``plc.tags`` too. 

3 - Not a programmed one.  The read method will look at the data type of what you are trying to read and will automatically break the list into as many requests as needed.  Part of uploading the tag list allows it to know how large the reply will be and account for that when building the requests.  If a tag is too big for a single packet, it will handle it separately in a fragmented request. 

4 - Thanks, it's pretty new so I appreciate the feedback.

AndrewStuhr, 2020-04-21:
Thanks!  

1. Yep, using :  'with LogixDriver(ip, large_packets=False) as plc:'  removed the error message

2. I was able to see the UDT structure by using 'plc.data_types['Equip0_Profile']'

3.  That's sweet

I'm going to start another topic regarding different questions.

closed 2020-04-22 by ottowayi