ottowayi/pycomm3 #32: Asyncronous Reads & Connection Timeout

AndrewStuhr, 2020-04-21:
1. Is there a way to read from multiple PLCs asynchronously?  For example, if I'm reading from 3 PLCs, can I send 3 read requests in quick succession and wait for them all to return, as opposed to reading each PLC individually and waiting for its response before reading the next PLC?

2. Do you have an suggestions for decreasing the connection timeout?  For example, only waiting 0.5 seconds before deeming the PLC inaccessible?  Also, related to question 1, if there is a way to asynchronously read, is there a way to set the timeout for all responses?

ottowayi, 2020-04-21:
1 - This library doesn't have async support, but you can use multithreading.  I use PyQt for my GUIs and typically just use QThreads for my clients and exchange signals when I need it to do something. 

2 - You can change the timeout on the socket after you open a connection. ``plc._sock.sock.settimeout(.5)`` But the CIP connection timeout is hard coded at the moment.  I set it to a pretty large value so that the PLC doesn't close the connection and prevent needing to send heart beats constantly to keep it active.   For now, if  you want to change it you'd need to change the values in the ``_forward_open`` method of the ``clx`` module.   It's been awhile since I looked at it, but I think the connection timeout is actually pretty complicated so I think I increased it and then didn't look at it much more.

closed 2020-04-22 by ottowayi

AndrewStuhr, 2020-07-22:
Hello again ottowayi!

When you mentioned multithreading, were you referring to the 'thread' library or 'multiprocessing' library?

Here's the situation I'm in.  I'm trying to read a somewhat arbitrary 1000 tags each second.  These tags are in the top level main program, for example 'Program:MainProgram.Bool1', 'Program:MainProgram.Float2', etc.  When reading 1000 tags using:

plc.read(*readArray)

it takes about 2 seconds - longer than I'd like.  I noticed that the time needed to read these tags increased proportionally with each 100 tags, so I'm assuming that's the limit per packet, and multiple packets get sent.  I tried to speed things up by creating a thread for each 100 tags to be able to read in parallel.  However, this still took the same amount of time.

Do you have any thoughts on speeding up reads, threading with pycomm3, or if my approach is crazy?  Thanks!

ottowayi, 2020-07-23:
I think what you're seeing is the time it takes for the driver to upload all of the tags (which it does automatically on connect).  I did a quick test:

```
if __name__ == '__main__':
    with LogixDriver('10.61.50.15/13') as plc:  # L85 direct
        atomic = [tag for tag in plc.tags if plc.tags[tag]['tag_type'] == 'atomic']
        count = len(atomic)
        start = time.time()
        plc.read(*(atomic*3))
        print(f'read {count* 3} tags in {time.time() - start} seconds')
```

`> read 1365 tags in 0.11809921264648438 seconds`

```
if __name__ == '__main__':
    with LogixDriver('10.61.50.228/13') as plc:  # L85 thru ENBT
        atomic = [tag for tag in plc.tags if plc.tags[tag]['tag_type'] == 'atomic']
        count = len(atomic)
        start = time.time()
        plc.read(*(atomic*3))
        print(f'read {count* 3} tags in {time.time() - start} seconds')
```

`> read 1365 tags in 0.2675485610961914 seconds`

Going thru an ENBT is much slower, but still only take a quarter of a second to read.  What does your code for reading these tags look like?  If you're creating a new driver every time you it will be much slower since it will upload the tags every time.  You can save the tag definitions from your first connection and pass them to your threads connections.  Just disable the `init_*` options when making the driver, then set `plc2._tags = plc1.tags`. That what I do when my apps were using the mutlithreading, but now I've removed it since adding the structure reading was a big enough of a performance boost I didn't need the multiple connections anymore.  Or, you could make a single long-last driver that is used repeatedly (just remember to call `plc.close()` when you're done).  Just one thing to remember, you need to keep the connection alive or it may be closed by the PLC after a while. (the CIP connection, the socket connection will still be open but your reads/writes will fail) I think what I do is just every 10s I read the PLC name if I haven't done any reading/writing.  Also, the instance ids are generated on a download. So if you do cache the tag definitions, you will have to reconnect to get the updated values after a download (or if new tags are added online).

AndrewStuhr, 2020-07-23:
Thanks for the response.  It still takes about 3 seconds to read 1,500 tags.  I removed the * 3.  It seems like it wasn't actually affecting the time needed to read tags even though there were more tags in the 'atomic' array.

[image]

AndrewStuhr, 2020-07-23:
This may just be a limitation of our PLC.  When using PyComm3 to read tags in a while loop on 2 separate computers, the speed was cut in half.  Let me know if you have any thoughts.

ottowayi, 2020-07-23:
Yeah I'm betting it's something like that.  Is there an HMI connected to it?  I've seen Kepware/IGS take up a lot of bandwidth and slow down other comms or in one case we had 4 panelviews running of the same CompactLogix and it really slowed down my app. It could also be the comms card too, and ENBT is much slower than an EN2T.  Also, you can try increasing the System Overhead Timeslice option in the PLC, I think Rockwell defaults them way too low for most applications.

AndrewStuhr, 2020-07-23:
There's not an HMI connected to it, but I'll try increasing the system overhead timeslice.  Standby...

AndrewStuhr, 2020-07-23:
The bottleneck was the PLC, not the PyComm3.  Changing the system overhead time slice proportionally changed our scan rate. 
 We're using an old L43 btw.  Thanks for all you're help man - you're the best!

ottowayi, 2020-07-24:
Awesome, glad that helped.