ottowayi/pycomm3 #33: Reading a Field Inside a UDT

AndrewStuhr, 2020-04-21:
I'm able to read contents inside a UDT:

ip 			= '192.168.50.29'
tagName 	= 'INT_Equip0_Step1_TMR'
with LogixDriver(ip, large_packets=False) as plc:
	print(plc.read(tagName))

[image]

The response shows that the timer has a CTR field.  However, when I use tagName = 'INT_Equip0_Step1_TMR', I get the following error:

[image]

I'm guessing this is a boneheaded move on my part, haha, but any ideas what's wrong?

AndrewStuhr, 2020-04-21:
Well I've narrowed down the issue a little.  I'm able to read the other fields PRE, ACC, EN, TT, and DN.  It seems that I'm having trouble reading CTL because it's a DWORD, which I believe is a 32 bit array.  I tried using tagName ='INT_Equip0_Step1_TMR.CTL{32}'  but that didn't work.

Any thoughts/advice on how to read from a DWORD?

ottowayi, 2020-04-21:
So pre-defined types get a little weird.  I'm not sure what that attribute is used for, but you will see most of the pre-defined types will have it.  But it's not shown in Logix, I'm guessing they allow it when reading the full structure but it's not accessible individually.  This is what the Data Access Manual says:

[image]

AndrewStuhr, 2020-04-22:
Gotcha. I'm not sure what that attribute is used for either, I was just poking around.

AndrewStuhr, 2020-04-22:
Are the LEN and DATA fields inside arrays another example of a field that can't be accesed individually?  I see these properties in data_types.internal_tags, but I get an error when I try to read them using <arrray_name>.LEN or <arrray_name>.DATA.

I'm able to read individual elements in the array <arrray_name>.[x] & I already know the array length, so this isn't really a big deal.  I just want to make sure I'm not missing something.

ottowayi, 2020-04-22:
Hmm I'm not sure exactly, could you post an example?

AndrewStuhr, 2020-04-22:
Here's my code:

from pycomm3 import LogixDriver

ip 			= '192.168.50.29'
tagName1 	= 'Internal_String_Array'
tagName2 	= 'Internal_String_Array[0]'
tagName3 	= 'Internal_String_Array.LEN'
with LogixDriver(ip, large_packets=False) as plc:
    print(plc.tags[tagName1]['data_type']['internal_tags'])
    print(plc.read(tagName2))
    print(plc.read(tagName3))

Here's the output:

[image]

There's 2 fields inside the array - DATA and LEN.  However, I get an error when I try to read them.

Readying from a particular element in the array works just fine though.

ottowayi, 2020-04-22:
Oh I see.  So the issue is you're missing the index when trying to read the attribute.  It should be ``tagName2 = 'Internal_String_Array[0].LEN'``.  The reason tagName1 and tagName2 are working is because by default the index is 0 and numbers of elements is 1.  But, to access the attributes you will need to specify the index in the request.  If you want to see how it works, check the ``packets/requests.py`` module and look at the ``_create_tag_rp`` method. 

I tested the different combos on a PLC of mine and this is what I get:

```    
with LogixDriver('10.61.50.4/10') as plc:
        print(plc.read('CIPStepDesc[0]', 'CIPStepDesc[0].LEN', 
                        'CIPStepDesc[0].DATA{82}', 'CIPStepDesc[10]{5}'))
```

```
[
Tag(tag='CIPStepDesc[0]', value='CIP Tank Fill', type='STRING', error=None),
Tag(tag='CIPStepDesc[0].LEN', value=13, type='DINT', error=None), 
Tag(tag='CIPStepDesc[0].DATA', value=[67, 73, 80, 32, 84, 97, 110, 107, 32, 70, 105, 108, 108, 
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 
type='SINT[82]', error=None), 
Tag(tag='CIPStepDesc[10]', value=['Fresh Water Rinse Charge to Circuit', '',
 'Fresh Water Rinse to Drain', 'Fresh Water Rinse to Recovery Tank',
 'Recovered Rinse to Drain'], type='STRING[5]', error=None)
]
```

I'm still working on fleshing out the [docs](https://pycomm3.readthedocs.io/en/latest/usage.html#reading-tags) with more examples, but there a couple in there regarding arrays.

closed 2020-04-29 by ottowayi