ottowayi/pycomm3 #35: Missing IO Tags

AndrewStuhr, 2020-04-29:
Hello.  I'm using the following code to read the tag structure of a PLC:
```
with LogixDriver(ipPlusSlot, large_packets=False, init_program_tags=True) as plc:
   result["tagStructure"] = plc.tags
```

plc.tags is returning 49 tag values.  However, when I use PyLogix's GetTagList() on the same PLC, I get 87 values back.  I believe the extra PyLogix tags are IO tags, such as "VFD_CV_313:I" and "IV:I".  Does PyComm have the ability to retrieve these types of tags?

Thanks

ottowayi, 2020-04-29:
So just took a look at how pylogix filters the tag list and it appears they only filter out a few specific things, which can lead to some garbage being left in the list.  I'm following the guidelines in the Data Access Manual from Rockwell (page 50, Eliminate tags by applying rules).  That being said, I could possibly add an option the the ``get_tag_list`` method that would not filter out Module-Defined tags. I'm on a startup right now, but if I get some time I will look into it.  If you want to try it out, look at the ``_isolating_user_tag`` method in the ``clx.py`` module. You can probably comment out the section that skips tags with a colon in them.

TODO: refactor ``_isolating_user_tag`` into ``_isolate_user_tags``

ottowayi, 2020-04-29:
So I looked into it a little more and create a branch called ``tag_list_extras`` if you want to try it out.  What I did was allow the base tags for each I/O module to be included in the list and then it also parses them into a dict included in the ``info`` property.  One thing I noticed was that any ``Local:*`` tags are not in the list, but the only PLCs I tested on had no local I/O modules so it could be that too. While I was at it I added the tasks/programs/routines to the info as well.  I may be able to do more with it in the future, but right now it's pretty much just for reference.

ottowayi, 2020-05-05:
Those changes have been merged into the ``develop`` branch.  They will be in the next release after I get some testing done on the other changes I've added.

ottowayi, 2020-05-11:
resolved in v0.6.0

closed 2020-05-11 by ottowayi

AndrewStuhr, 2020-05-12:
Hey I was finally able to test this out, and it worked!  Thanks a bunch