ottowayi/pycomm3 #36: Writing STRING array fails above 4000 bytes

MikeDausch, 2020-05-04:
Only 45 strings can be in a write before the exception.
"pycomm3.RequestError: Fragmented write of structures is not supported"
This falls in the "struct" tag_type and "Write_tag_fragmented"
It would be great to have this fixed. Thank you.

ottowayi, 2020-05-05:
Currently, I don't support writing of structures except for strings and I must have overlooked the possibility of writing a large array of them.  My latest [commit](https://github.com/ottowayi/pycomm3/commit/4cbbd512d012bcd5c24f510b382e114782c5ba4a) to ``develop`` should resolve it, would you be able to help test it?   All of the changes are in the [WriteTagFragmentedServiceRequestPacket](https://github.com/ottowayi/pycomm3/blob/4cbbd512d012bcd5c24f510b382e114782c5ba4a/pycomm3/packets/requests.py#L277)  in ``pycomm3/packets/requests.py`` line 277.  You should just be able to copy and paste the whole class into your version to get it working.  I'm currently working on a few other improvements as well, but it will be a bit before I can get a new version released.

MikeDausch, 2020-05-06:
Thank you very much for taking a look at this.  Yes I will definitely do
the test.  I can help in anyway.  I have access to many PLCs too.
Im a novice when it comes to Python but I'll give it a try.
Mike Deuschle

ottowayi, 2020-05-06:
Awesome, thanks.  You could also update to the develop branch by doing `` pip install --upgrade git+git://github.com/ottowayi/pycomm3.git@develop``  if you don't want to mess with editing the library.  Feel free to ask if you need any help or just have questions.

MikeDausch, 2020-05-06:
wow.
That definitely works. Thank you.
I have a large text file that I need to send as separate lines to a String
Array.

plc2.write(('PART_PROGRAMS[0,0]{'+str(length)+'}', s_lines))

it takes
10.0975 sec to transfer 401 strings.
How do I get a progress bar to update during the transfer?

Mike Deuschle

ottowayi, 2020-05-06:
I found a bug in how I was breaking up the write data into the fragments, so I was sending way more requests that it should have taken. That's why you were seeing it take so long. I just ran this test over a VPN back to my office:

```
    with LogixDriver('10.61.50.4/10') as plc:
        num = 500
        lines = [f'Test line {i} out of {num}.' for i in range(num)]
        start = time()
        plc.write((f'TestStrings{{{num}}}', lines))
        end = time()

        print(f'Total time to write {num} strings: {end-start} seconds.')

Total time to write 500 strings: 0.25632286071777344 seconds.
```

Sorry about that, you should be able to re-run that update command and get the fix.  Also, there isn't really anyway to show a progress bar unless you customize some of the classes.  But, you can enable the logging so you can see what it is doing like:

```
import logging
logging.basicConfig(level=logging.DEBUG, stream=sys.stdout, force=True)
```

which will get you output like:

```
...
DEBUG:pycomm3.packets.requests.WriteTagFragmentedServiceRequestPacket:Sent: WriteTagFragmentedServiceRequestPacket() (part=12 offset=43780)
DEBUG:pycomm3.packets.requests.WriteTagFragmentedServiceRequestPacket:Received: WriteTagFragmentedServiceResponsePacket(service=b'\xd3', command=b'p\x00', error=None)
DEBUG:pycomm3.packets.requests.WriteTagFragmentedServiceRequestPacket:Reassembled Response: WriteTagFragmentedServiceResponsePacket(service=b'\xd3', command=b'p\x00', error=None)
Total time to write 500 strings: 0.2569413185119629 seconds.
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: SendRRDataRequestPacket()
DEBUG:pycomm3.packets.requests.RequestPacket:Received: SendRRDataResponsePacket(service=b'\xce', command=b'o\x00', error=None)
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: UnRegisterSessionRequestPacket()
DEBUG:pycomm3.packets.requests.RequestPacket:Received: UnRegisterSessionResponsePacket()
```

MikeDausch, 2020-05-06:
Amazing.  Yes.  If the transfer time is less than a sec then there is no
need for the progress bar.
401 strings in 0.1904904842376709 seconds

Great work.
I didn't even realize  the 10 sec transfer was a bug.

Thank you,
Mike Deuschle

ottowayi, 2020-05-06:
Thanks, glad to hear it's working.  I should be rolling out a new version one of these days once I get some tests written and stuff. If you want to be notified you can just click that 'Watch' button on the top of the page.

closed 2020-05-06 by MikeDausch