ottowayi/pycomm3 #39: Question about Array Structure in plc.tags

AndrewStuhr, 2020-05-12:
When should I look at 'dim' & 'dimensions' to determine if a tag is an array, and when should I look at 'array' to determine if a tag is an array?  

Here's a picture of a particular tag in plc.tag.  It has 'dim' & 'dimensions' attributes, but not 'array'.  It's child 'RecStage' has the 'array' attribute, but not 'dim' nor 'dimensions'.

[image]

ottowayi, 2020-05-13:
Hmm, yeah that is sort of an inconsistency in naming.   ``dim`` is the number of dimensions, 0 not an array, 1/2/3 the number of dimensions.  ``dimensions`` has the size of each dimension.  So a DINT[50] would have ``dim: 1`` and ``dimensions:[50, 0, 0]``, and DINT[10, 20, 30] would be ``dim:3`` and ``dimensions:[10, 20, 30]``.  ``array`` is only for members of structures since you are limited to 1D arrays inside UDTs/AOIs/etc in Logix.  I think I used the different names based on how they are described in the Rockwell manual, but I'm not opposed to making them the same name either.  The only thing is for base tags it will always be an 3-element list and struct members will just be a int.

AndrewStuhr, 2020-05-15:
Thanks!  I must have missed the notification when you posted this.  I just looked at the docs - is this basically correct:

If a tag has the 'internal_tags' property', it's a struct and use 'array'.  Otherwise, it's not a structure and use 'dim'/'dimensions'?

Your response makes sense, just want to make sure I fully understand.  I'm thinking I'll use the logic above in my javascript to determine which property to use.

ottowayi, 2020-05-15:
Sort of, you'll probably have to do it in 2 steps. All top level tags will have the dim attributes. So you can use that to determine what base tags are arrays. Then look for `tag_type: 'struct'` base tags, their `data_type` attribute will be a dict that you can look for the `array` attribute for it's internal tags. Hope that makes sense, sorry I'm on mobile else I would give more concrete examples.

AndrewStuhr, 2020-05-15:
No worries, yea that helps.  I'll give it a try.  Thanks!

closed 2020-05-26 by ottowayi