ottowayi/pycomm3 #40: Queston: Program Scoped Tags

coltman151, 2020-05-17:
Is there a way to handle program-scoped tags? I've tried adding "Program:{myprogram}.{tagtoread}" but I only get an invalid tag request from it.

coltman151, 2020-05-17:
I figured it out, it's on the docs website. You must set LogixDriver init_program_tags=True before calling plc.read('Program:{desiredprogram}.{tagtoread}.

ottowayi, 2020-05-18:
I'm reopening this because there may be an issue, you should not need the separate call to get_tag_list, just the init arg.

ottowayi, 2020-05-18:
I just double checked and it works with only ``init_program_tags=True``.  You will need to use the format of ``'Program:{program_name}.{tag_name}'``, but it shouldn't be any different than using program tags in FTView or other HMIs. I added a few tests for them too.  Feel free to re-open this issue if you're still having to call ``get_tag_list`` separately, but you shouldn't need to.

closed 2020-05-18 by ottowayi

coltman151, 2020-05-18:
Okay I'll give it a try again when I'm back in front of a PLC. I'm pretty weak with python, so it very well could have been me.

souirite, 2021-04-15:
I tried using the method you've stated above and I'm still having issues.
This is how my code is structured:
plc = LogixDriver('136.13.0.11/2', init_program_tags=True)
plc.open()
program_name ='S02_StnL010_TT'
tag_name = 'zzTypeIndex'
print(plc.read(f'Program:{program_name}.{tag_name}'))
plc.close()

The printed is result is this:
Program:S02_StnL010_TT.zzTypeIndex, None, None, Destination unknown, class unsupported, instance undefined or structure element undefined (see extended status) - Extended status out of memory  (05, 00)

ottowayi, 2021-04-15:
Can you try adding `plc._cfg['use_instance_ids'] = False` before reading the tag.  If that works, I have a bug.  If not, then something else is wrong.  I have a feeling that using instance ids for program tags doesn't work, but I have no hardware that I can test that on.  My only Logix device is v20, v21 was when instance id support was added.

slowMelon01, 2021-04-15:
I am working with @souirite on this and I've tried adding `plc._cfg['use_instance_ids'] = False` and it worked.
Code:
`plc = LogixDriver('136.13.0.11/2', init_program_tags=True)`
`plc.open()`
`program_name ='S02_StnL010_TT'`
`tag_name = 'zzTypeIndex'`
`plc._cfg['use_instance_ids'] = False`
`print(plc.read(f'Program:{program_name}.{tag_name}'))`
`plc.close()`

Output:
Program:S02_StnL010_TT.zzTypeIndex, 1, DINT, None

If it helps, here is the hardware information we are using:
vendor: Rockwell Automation/Allen-Bradley
product_type: Programmable Logic Controller
product_code: 213
revision: {'major': 31, 'minor': 12}
status: b'`1'
serial: 01105868
product_name: 1756-L83ES/B
keyswitch: REMOTE RUN

ottowayi, 2021-04-15:
Well fuck. I wish I would have looked at your comment before I published the latest release literally 5 minutes beforehand.  Oh well, I'll try and get a fix out today.

ottowayi, 2021-04-15:
If you can try making a small code change an verify that it fixes the issue, I can publish a new release. 

In `pycomm3/packets/util.py` line 78 inside the `tag_request_path` function, this condition needs to be added:
[image]

If that works for you without setting the `use_instance_ids` setting, I can get a new release out right away.  I already confirmed it doesn't break anything with my PLC.

slowMelon01, 2021-04-15:
I will look into this tomorrow morning.

ottowayi, 2021-04-15:
Awesome thanks.

slowMelon01, 2021-04-16:
I've added the line of code you asked and it still works when I removed the `plc._cfg['use_instance_ids'] = False` code before I read the tag.

Cheers for your help.

ottowayi, 2021-04-16:
Great, thanks for doing that.  I just published the new release with the fix (1.1.1).