ottowayi/pycomm3 #42: STRING array write stops at 361 (0-360)

MikeDausch, 2020-05-19:
Attempting to write 401 STRINGs stops at 360 element.
I believe this used to work. I get no errors.
Thank you for looking at this.

ottowayi, 2020-05-19:
Hmm, I just tried it on a v24 L73 with no issues: 

[image]

Using this code:

    with LogixDriver('10.61.50.4/10') as plc:
        num = 500
        lines = [f'Test line {i} out of {num}.' for i in range(num)]
        plc.write((f'TestStrings{{{num}}}', lines))

Could you share your code and maybe some information on about the PLC? Also, you can confirm you're running the latest version with the command: ``pip show pycomm3``.  My email is also listed in there, so you can email me if you don't want to post anything publicly.

ottowayi, 2020-05-19:
The docs state that the value list will be truncated to the limit in the request (the '{x}') if it is too long, this does not appear to be the case. It attempts to write beyond that limit and past the bounds of the array.

ottowayi, 2020-05-22:
fixed in version 0.6.2

closed 2020-05-22 by ottowayi