ottowayi/pycomm3 #52: Can't read some structures - "Permission denied - No Extended Status"

ambersnow, 2020-06-19:
Hi,

I've been using this library for a bit and it has been amazing upgrade to the other ones I've used in the past. I am enjoying the structure reading but I have noticed that on some structures it fails with an permission denied - no extended status while on others (eg, TIMER) it reads all the attributes just fine. 

I can get around this by using the datatypes collected by pycomm3 and manually explode the tag into its attributes and reading that way but I was curious as to what causes the "permission denied - no extended status".

```
>>> plc.read("M_1001.FTS")
Tag(tag='M_1001.FTS', value=None, type=None, error='Permission denied - No Extended Status')
>>> plc.read("M_1001.HMI_START") 
Tag(tag='M_1001.HMI_START', value={'Control': 0, 'PRE': 2000, 'ACC': 0, 'EN': False, 'TT': False, 'DN': False}, type='TIMER', error=None)
```

HMI_START is a TIMER whilst FTS is an ALM_DIG (which is an add-on instruction). Like I said I can get around it by manually reading each individual attribute but I was curious if there is something misconfigured or something to look for as to why it fails on some structures but not others.

Thank you!

ottowayi, 2020-06-19:
Look in your AOI definition, if there are any tags that have an external access of None it will give you that error. Unfortunately there is no work around besides reading the individual attributes.  If it's an AOI you control, setting those to Read Only should resolve it.  I recently lobbied and got our AOIs updated to change any attributes with None to Read Only since reading performance is much better when you can do a whole struct at once.

ambersnow, 2020-06-19:
Ah!!!! That makes sense, turns out 2 of the attributes were set to None which I had also noticed when I was reading each attribute manually and only two of them were reading nothing.

Thank you!

ottowayi, 2020-06-19:
No problem. You should also see those 2 attributes are NOT in the data type attribute list either.  So to anything other than the PLC/Logix, they don't exist.

ambersnow, 2020-06-19:
They are in the datatype attribute list for that AOI -- which is why I was curious as to why it wasn't able to read the entire structure. They appear in the internal_tags object along with the attributes list for the ALM_DIG AOI.

ambersnow, 2020-06-19:
Which brings up an idea -- is it possible for pycomm3 to know the external_access on datatype attributes like it can on the tags specifically?

ottowayi, 2020-06-19:
Sorry I was mistaken and commented without double checking, I think it's InOut parameters that don't show up.  I was able to find the external access attribute in the Rockwell manual for [tags](https://pycomm3.readthedocs.io/en/latest/usage.html#tag-structure) , but wasn't able to find anything about structure attributes.  I remember doing some trial and error to see if I could get it while getting all of the other structure information, but I never did.  I would like to include it if it's possible, but there is a lot Rockwell doesn't document.

ambersnow, 2020-06-19:
Sounds like typical Rockwell. Thanks for the info!

closed 2020-06-19 by ambersnow

ottowayi, 2020-06-19:
No problem, it's still on my list of things I would like to add (assuming it's possible).  It's only 2-bits and there is attribute 6 for the Symbol Object that's description is just 'storage for whatever'.  It may be in there, I just haven't had the time to test it out yet.  But I think one of the bits in there is an alias flag too, which is where I get my 'alias' attribute from.

ambersnow, 2020-06-19:
Yeah having the access to that rw/ro/none access on the datatype would be super useful but in the time being I can attempt to read the attributes and if it fails then consider that as "no access". I can totally help test stuff out if needed