ottowayi/pycomm3 #54: Instance Addressing

AndrewStuhr, 2020-06-26:
When pycomm3 reads a tag, such as:

plc.read('DINT1')

does it send the tag name 'DINT1' to the PLC, or does it send 'DINT1's instance ID?

I'm curious because recently I learned that Kepware can send the tag address instead of the tag name during read requests.  This increases the amount of data that can fit in a packet and vastly speed up reads.

Thanks!

ottowayi, 2020-06-26:
Yes it does by default, part of the initial connection is requesting the identity info for the PLC and if you're using a firmware that doesn't supports it (<21 or a Micro800) it will disable it.  Of course, if you disable that with the `init_info=False` option it requires setting the `use_instance_ids` attribute to False or the requests will fail.

AndrewStuhr, 2020-06-26:
Just out of curiousity, do you have any idea how many addresses can fit in a read request using the instance ids?

ottowayi, 2020-06-26:
I'm on my phone right now so can't check, but the IDs are only a few bytes. So depending on if you're using a standard forward open or extended you have 500 or 4000 bytes of request data to work with. It will automatically try the extended one first and fall back to the standard if it fails. I think it's v20+ and an EN2T or newer is required for the extended. Also if you're using kepware, you can increase the size to 4000 instead of the standard 500 (I think it's in Logix options).  One thing that pycomm3 does though is track how large your response will be, so it makes sure to only pack enough requests to fit in the reply. So you can fit more DINT tags in a request than say structures that are 500 bytes a piece.

closed 2020-07-09 by ottowayi