ottowayi/pycomm3 #58: [BUG] - Boolean Read fails when value is True (Succeeds when false)

alecjohanson, 2020-07-20:
**Pre-checks**
- [x ] running latest version (`pip show pycomm3`)
Name: pycomm3
Version: 0.8.1
Summary: A Python Ethernet/IP library for communicating with Allen-Bradley PLCs.
Home-page: https://github.com/ottowayi/pycomm3
Author: Ian Ottoway
Author-email: ian@ottoway.dev
License: MIT
Location: /home/vagrant/.local/lib/python3.6/site-packages
Requires: autologging
- [?] enabled logging 
- [x] consulted the [docs](https://pycomm3.readthedocs.io/en/latest/) 

**Description**
_A clear and concise description of what the bug is._

**Target PLC**
Model:  [Micro820]
Firmware Revision: [11]
Other Devices in CIP Path:  []

**Code Sample**
_Minimal reproduceable code sample_
```python
with LogixDriver("192.169.10.7") as plc:
     print(plc.read("Data_out_control"))
```

**Additional context**
 - Reading a tag that is Boolean type

alecjohanson, 2020-07-20:
A successful read when the value is False/'0':
```

>>> from pycomm3 import LogixDriver
>>> with LogixDriver("192.169.10.7") as plc:
...     print(plc.read("Data_out_control"))
...
Data_out_control, False, BOOL, None
```

alecjohanson, 2020-07-20:
An unsuccessful read when the value it True/1:

`>>> with LogixDriver("192.169.10.7") as plc:
...     print(plc.read("Data_out_control"))
...
Traceback (most recent call last):
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/packets/requests.py", line 86, in _build_header
    Pack.udint(self._plc._session),  # Session Handle UDINT
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/bytes_.py", line 67, in <lambda>
    udint = lambda n: pack('<I', n)
struct.error: argument out of range
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/clx.py", line 170, in __init__
    self._micro800 = self._list_identity().startswith(MICRO800_PREFIX)
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/clx.py", line 319, in _list_identity
    response = request.send()
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/packets/requests.py", line 137, in send
    self._send(self._build_request())
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/packets/requests.py", line 72, in _build_request
    header = self._build_header(self._encap_command, len(msg))
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/packets/requests.py", line 93, in _build_header
    raise CommError(e)
pycomm3.CommError: argument out of range
`

alecjohanson, 2020-07-20:
Strangely, Setting the logger to DEBUG seems to fix the issue...

```
>>> logging.getLogger(__name__).setLevel("DEBUG")
>>> with LogixDriver("192.169.10.7") as plc:
...     print(plc.read("Data_out_control"))
...
Data_out_control, True, BOOL, None
```

alecjohanson, 2020-07-20:
It appears that this will sometimes succeed, but doesn't appear to be deterministically.

ottowayi, 2020-07-20:
Hmm, I don't think it's actually related to the value of that tag.  That exception is thrown before it even opens the connection, I'm wondering if it is related to the bug I fixed this morning.  The session handle was being unpacked as a DINT, not a UDINT - so when going to pack a negative it would be out of range.  It could just be coincidence for when it's working or not working.  

Can you try editing [this](https://github.com/ottowayi/pycomm3/blob/6622c7df5cf976817aefc7f44d86adad45c8a700/pycomm3/packets/responses.py#L345) line to be ``self.session = Unpack.udint(self.raw[4:8])`` and see if that works?  I just committed it to the dev branch this morning, so it hasn't made it to master/been released yet.

alecjohanson, 2020-07-22:
I still see this error with that change. 

FWIW: I cloned this repo, made the change and changed __version__. I used ```sudo python3 setup.py install``` and verified with pip3 list that the new version was the one being executed.

ottowayi, 2020-07-22:
Interesting, I've never this issue before although I don't use Micro800s, so they may act a little differently.  I had another user with a similar issue that made me think it was the same since both only happened with a Micro800.  I've only used the simulator to test since my company only uses the Logix platforms.  Could you enable the logging and attach them?  There is a logging section in the docs, see the part on the VERBOSE_DEBUG option.  That will help a lot since I can see what it is sending/receiving.

alecjohanson, 2020-07-26:
Here is the verbose output of a failed read:

```
DEBUG:pycomm3.packets.requests.RequestPacket:>>> SEND >>>
(0000) 65 00 04 00 00 00 00 00 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 01 00 00 00
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: RegisterSessionRequestPacket(message=[b'\x01\x00', b'\x00\x00'])
DEBUG:pycomm3.packets.requests.RequestPacket:<<< RECEIVE <<<
(0000) 65 00 04 00 5f f7 6f ad 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 01 00 00 00
DEBUG:pycomm3.packets.requests.RequestPacket:Received: RegisterSessionResponsePacket(session=-1385171105, error=None)
DEBUG:pycomm3.clx.LogixDriver:Session = -1385171105 has been registered.
Traceback (most recent call last):
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/packets/requests.py", line 86, in _build_header
    Pack.udint(self._plc._session),  # Session Handle UDINT
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/bytes_.py", line 67, in <lambda>
    udint = lambda n: pack('<I', n)
struct.error: argument out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 9, in <module>
    with LogixDriver("192.169.10.7") as plc:
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/clx.py", line 170, in __init__
    self._micro800 = self._list_identity().startswith(MICRO800_PREFIX)
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/clx.py", line 319, in _list_identity
    response = request.send()
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/packets/requests.py", line 137, in send
    self._send(self._build_request())
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/packets/requests.py", line 72, in _build_request
    header = self._build_header(self._encap_command, len(msg))
  File "/home/vagrant/.local/lib/python3.6/site-packages/pycomm3/packets/requests.py", line 93, in _build_header
    raise CommError(e)
pycomm3.CommError: argument out of range
```

alecjohanson, 2020-07-26:
Compared to a successful read: 

```
DEBUG:pycomm3.packets.requests.RequestPacket:>>> SEND >>>
(0000) 65 00 04 00 00 00 00 00 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 01 00 00 00
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: RegisterSessionRequestPacket(message=[b'\x01\x00', b'\x00\x00'])
DEBUG:pycomm3.packets.requests.RequestPacket:<<< RECEIVE <<<
(0000) 65 00 04 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 01 00 00 00
DEBUG:pycomm3.packets.requests.RequestPacket:Received: RegisterSessionResponsePacket(session=1834148296, error=None)
DEBUG:pycomm3.clx.LogixDriver:Session = 1834148296 has been registered.
DEBUG:pycomm3.packets.requests.RequestPacket:>>> SEND >>>
(0000) 63 00 00 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: ListIdentityRequestPacket(message=[])
DEBUG:pycomm3.packets.requests.RequestPacket:<<< RECEIVE <<<
(0000) 63 00 37 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 01 00 0c 00 31 00
(0030) 01 00 00 02 af 12 c0 a9 0a 07
(0040) 00 00 00 00 00 00 00 00 01 00
(0050) 0e 00 b4 00 0b 0b 34 00 9e 29
(0060) c5 60 0f 32 30 38 30 2d 4c 43
(0070) 32 30 2d 32 30 51 57 42 03
DEBUG:pycomm3.packets.requests.RequestPacket:Received: ListIdentityResponsePacket(identity='2080-LC20-20QWB', error=None)
DEBUG:pycomm3.packets.requests.RequestPacket:>>> SEND >>>
(0000) 6f 00 1a 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 00 00 00 00 0a 00
(0030) 02 00 00 00 00 00 b2 00 0a 00
(0040) 01 02 20 01 24 01 01 00 01 00
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: GenericUnconnectedRequestPacket(message=[b'\x01', b'\x02 \x01$\x01', b'', b'\x01\x00\x01\x00'])
DEBUG:pycomm3.packets.requests.RequestPacket:<<< RECEIVE <<<
(0000) 6f 00 3d 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 00 00 00 00 05 00
(0030) 02 00 00 00 00 00 b2 00 2d 00
(0040) 81 00 00 00 01 00 0e 00 b4 00
(0050) 0b 0b 34 00 9e 29 c5 60 0f 32
(0060) 30 38 30 2d 4c 43 32 30 2d 32
(0070) 30 51 57 42 03 00 00 00 65 6e
(0080) 67 65 6e 67 00
DEBUG:pycomm3.packets.requests.RequestPacket:Received: GenericUnconnectedResponsePacket(service=None, command=b'o\x00', error=None)
DEBUG:pycomm3.packets.requests.RequestPacket:>>> SEND >>>
(0000) 6f 00 42 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 00 00 00 00 0a 00
(0030) 02 00 00 00 00 00 b2 00 32 00
(0040) 5b 02 20 06 24 01 0a 05 00 00
(0050) 00 00 9a 9e 0f ec 27 04 09 10
(0060) 89 db b0 e5 07 00 00 00 01 40
(0070) 20 00 a0 0f 00 42 01 40 20 00
(0080) a0 0f 00 42 a3 02 20 02 24 01
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: GenericUnconnectedRequestPacket(message=[b'[', b'\x02 \x06$\x01', b"\n\x05\x00\...\x0f\x00B\xa3", b'\x02 \x02$\x01'])
DEBUG:pycomm3.packets.requests.RequestPacket:<<< RECEIVE <<<
(0000) 6f 00 2e 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 00 00 00 00 05 00
(0030) 02 00 00 00 00 00 b2 00 1e 00
(0040) db 00 00 00 1a 0e f2 31 9a 9e
(0050) 0f ec 27 04 09 10 89 db b0 e5
(0060) e8 3d 20 00 e8 3d 20 00 00 00
DEBUG:pycomm3.packets.requests.RequestPacket:Received: GenericUnconnectedResponsePacket(service=None, command=b'o\x00', error=None)
DEBUG:pycomm3.packets.requests.RequestPacket:>>> SEND >>>
(0000) 70 00 2c 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
[... 178 lines cut ...]
Data_out_control, False, BOOL, None
DEBUG:pycomm3.packets.requests.RequestPacket:>>> SEND >>>
(0000) 6f 00 26 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 00 00 00 00 0a 00
(0030) 02 00 00 00 00 00 b2 00 16 00
(0040) 4e 02 20 06 24 01 0a 05 27 04
(0050) 09 10 89 db b0 e5 02 00 20 02
(0060) 24 01
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: GenericUnconnectedRequestPacket(message=[b'N', b'\x02 \x06$\x01', b"\n\x05'\x04...9\xdb\xb0\xe5", b'\x02\x00 \x02$\x01'])
DEBUG:pycomm3.packets.requests.RequestPacket:<<< RECEIVE <<<
(0000) 6f 00 1e 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00 00 00 00 00 05 00
(0030) 02 00 00 00 00 00 b2 00 0e 00
(0040) ce 00 00 00 27 04 09 10 89 db
(0050) b0 e5 00 00
DEBUG:pycomm3.packets.requests.RequestPacket:Received: GenericUnconnectedResponsePacket(service=b'N', command=b'o\x00', error=None)
DEBUG:pycomm3.packets.requests.RequestPacket:>>> SEND >>>
(0000) 66 00 00 00 c8 e1 52 6d 00 00
(0010) 00 00 5f 70 79 63 6f 6d 6d 5f
(0020) 00 00 00 00
DEBUG:pycomm3.packets.requests.RequestPacket:Sent: UnRegisterSessionRequestPacket(message=[])
DEBUG:pycomm3.packets.requests.RequestPacket:Received: UnRegisterSessionResponsePacket()
```

alecjohanson, 2020-07-26:
Looking at the debug output. It does seem to be a similar issue to what you were saying I think. It appears it's erroring out on getting the session ID, which seems to be in the first word of the response (with endianness changed).

Would you agree?

ottowayi, 2020-07-27:
Hmm so it looks like it's unpacking it signed, when it should be unsigned.  Try updating to v0.9.0 that I just posted and if that doesn't fix it let me know.

alecjohanson, 2020-07-28:
v0.9.0 seems to fix it...must have done something wrong when trying the same change locally.

closed 2020-07-28 by alecjohanson

ottowayi, 2020-07-28:
No worries, as long as it's working for you that's all that matters. Don't hesitate to open issues either, I don't typically work with Micro800s so opening issues helps me a lot in finding things that are different or broken when compared to the Logix PLCs.

alecjohanson, 2020-07-28:
Will do. Thanks for taking the time to look at this.