ottowayi/pycomm3 #62: disable debug logging

mytalala, 2020-08-05:
i want to disable debug level logging, but keep errors. i tries:

`
logging.getLogger("pycomm3").setLevel(logging.ERROR)
`

but it doesn't do anything. i use same line to disable _requests_ lib logging and its works fine

ottowayi, 2020-08-05:
That was something that I broke and didn't notice when I removed the autologging dependency.  I only had the __qualname__ for the logger name, so you would have to set the level on every class logger.  I was missing the module name, so `pycomm3` wasn't the root logger for the library anymore.  This should be fixed now in v0.9.2.

mytalala, 2020-08-05:
nice! i will wait for v0.9.2
thanks

closed 2020-08-05 by mytalala

ottowayi, 2020-08-05:
No problem, it was a mistake on my part.  I published it so you should be able to update anytime