ottowayi/pycomm3 #66: CIP Generic Message Instance Object Does not Exist

coltman151, 2020-08-08:
**Pre-checks**
- [x] running latest version (`pip show pycomm3`)
- [ ] enabled logging 
- [x] consulted the [docs](https://pycomm3.readthedocs.io/en/latest/) 

**Description**
I'm going to report this as a bug, although I'm not sure if it's necessarily a bug with _your_ programming or the way CIP Generic Messaging works. I'm building an app to get parameters from a PF525. If I try to get an instance greater than 255 (hex: FF), I get a "Object does not exist" error back. I'm almost positive it is because once you're above 255 in hexadecimal, it's 3 characters. 

example:

```drive.generic_message(
    service=CommonService.get_attribute_single,
    class_code=b'\x93'
    instance=b'\x1F'
    ...
```

will return Motor NP Volts no problem. However,

```drive.generic_message(
    service=CommonService.get_attribute_single,
    class_code=b'\x93'
    instance=b'\x216'
    ...
```

Returns the "Object Does Not Exist" error. Note that this _does_ work when setting up a MSG instruction in ladder, which is why I decided to report it as a bug. 

**Target Devices**
Model: 1756-L71, 25B-D2P3N104
Firmware Revision: 24.11, 6.001

ottowayi, 2020-08-09:
I think the issue is because I decided to make the args all just raw bytes and the MESSAGE configure dialog takes the (hex) integer values.  The integer value of `b'\x216'` is actually  `13857`, which is out of the range of parameters for the drive.  

Can you try doing this:

```
from pycomm3 import Pack
...
    class_code=b'\x93',
    instance=Pack.uint(216)
```

I think it would be a good idea to allow it to accept either ints or bytes for those, since it is kinda confusing.

coltman151, 2020-08-09:
I will try that in just a second. I actually just set up a breakpoint on where it packs the instance, and it's showing the value of instance is b'!6' when it makes the packet to send, but b'x\216' when being passed to the module. 

Being able to accept ints would be pretty handy, as converting to hex on the fly was a huge pain. Right now I'm passing the hex value as a byte string directly from a dictionary rather than trying to convert from an integer to hex and get everything to line up.

coltman151, 2020-08-09:
That fixes it, and actually simplifies passing instances to the module significantly. Simply send the parameter number and everything comes back correctly.

ottowayi, 2020-08-09:
Awesome, glad that worked.  I just pushed to the develop branch a change that lets it accept either bytes or ints if you want to try that out.  I don't have a drive to test against, but it passed all the tests and also worked when I did a get_plc_time w/ using `0x01` instead of `b'\x01'` for the instance value. If it ends up working well I'll put I'll put it in the next release.

coltman151, 2020-08-09:
Once I figure out how to install the development branch I can test it. I have a test stand with a whole bunch of EIP devices in front of me right now.

ottowayi, 2020-08-09:
Try using this to install it: `pip install git+https://github.com/ottowayi/pycomm3@develop`

And yeah, that would be great if you can do some testing.  I've been meaning to but been pretty busy with project work recently and don't really have much for non-PLC hardware.  If you find anything that you think would be useful for others, I'd definitely be willing to add them to the examples and stuff.

coltman151, 2020-08-09:
Works great passing an INT directly to instance with a PF525. Hopefully next weekend I'll have time to start trying generic messaging with some of the other devices I have laying around. 

Once I get a more up to speed on Python I'd like to start contributing a little more, but for the time being if you have anything else you want tested, I have resources to do that.

ottowayi, 2020-08-09:
Great!  I'll get that into the next update then.  And yeah, contributions are great.  I can't think of anything at the moment, but examples would definitely be a place to a start.  Even just code snippets, I can always add them to the docs myself.  Also, feel free to email me (ian@ottoway.dev).

ottowayi, 2020-08-11:
completed in #67

closed 2020-08-11 by ottowayi