ottowayi/pycomm3 #68: [BUG] - get_module_info attempts to concatenate 'int' and 'bytes'.

alphabet5, 2020-08-12:
**Pre-checks**
- [ ] running latest version (`pip show pycomm3`)

- [ ] enabled logging 
- [ ] consulted the [docs](https://pycomm3.readthedocs.io/en/latest/) 

**Description**
_A clear and concise description of what the bug is._

The function get_module_info attempts to concatenate 'int' and 'bytes'. It's not clear what the format for 'slot' should be. It would be helpful to have function help included in the code so that help(pycomm3.clx.LogixDriver.get_module_info) displays more useful information.

I've tried tracing this down, but I'm unclear what is being attempted / the purpose of this line in the function. Pack.sint(slot) with sint = lambda n: pack('b', n) appears to require an integer.

**Target PLC**
Model:  1756-L73S/B LOGIX5573SAFETY
Firmware Revision: 28.12
Other Devices in CIP Path: 1756-EN2T

**Code Sample**
_Minimal reproduceable code sample_
```python
 plc.get_module_info(1)
```

**Additional context**
 - _if reading/writing a tag, describe it's type/structure/etc_
 - _attach any relevant L5X or data files_
 - _paste/attach logging output_

```python
plc.get_module_info(1)
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pycomm3/cip_base.py", line 224, in get_module_info
    route_path=PATH_SEGMENTS['bp'] + Pack.sint(slot)
TypeError: unsupported operand type(s) for +: 'int' and 'bytes'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/IPython/core/interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-4-74a1cabf58cc>", line 1, in <module>
    plc.get_module_info(1)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pycomm3/cip_base.py", line 233, in get_module_info
    raise DataError('error sending request') from err
pycomm3.exceptions.DataError: error sending request
plc.get_module_info(0)
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pycomm3/cip_base.py", line 224, in get_module_info
    route_path=PATH_SEGMENTS['bp'] + Pack.sint(slot)
TypeError: unsupported operand type(s) for +: 'int' and 'bytes'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/IPython/core/interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-5-31842fcba601>", line 1, in <module>
    plc.get_module_info(0)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pycomm3/cip_base.py", line 233, in get_module_info
    raise DataError('error sending request') from err
pycomm3.exceptions.DataError: error sending request
```

ottowayi, 2020-08-12:
Yeah that's a bug... and I should have a test for it and caught that.  The `PATH_SEGMENTS['bp']` should be `Pack.usint(PATH_SEGMENTS['bp'])`.  I'll get it fixed tomorrow, probably going to change that constant to be already in bytes.

ottowayi, 2020-08-12:
fixed in #69

closed 2020-08-12 by ottowayi