ottowayi/pycomm3 #70: Program Tags using init_program_tags

AndrewStuhr, 2020-08-12:
Hello

I'm trying to scan program tags from a plc, but this isn't returning them.  I'm probably doing something dumb, but just wondering if you see an issue with this code.  Thanks!

from pycomm3 import LogixDriver

ipAndSlot = '10.11.12.218/0'
try:
    plc = LogixDriver(ipAndSlot, init_program_tags=True)
    print(plc.get_tag_list())
    
except Exception as e:
    print(str(e))

ottowayi, 2020-08-12:
You're not doing anything _dumb_, just not doing what you're thinking you are.  In the initializer, it will call `get_tag_list` automatically if the `init_tags` arg is True (the default), by setting `init_program_tags` it tells that `get_tag_list` call include the program-scoped tags.  But, in your second explicit call to `get_tag_list`, it's optional argument `program` is defaulted to None.. which means controller scoped only.  If you want to get all program scoped tags, you'd do `plc.get_tag_list(program='*')`.  Or to just get the tags from a single program you'd do `plc.get_tag_list(program='MyProgram')`.

But, you don't even need to call `get_tag_list` since it is called in the initializer automatically and stores the tag list in the `tags` property. 

So you could do:
```
from pycomm3 import LogixDriver

ipAndSlot = '10.11.12.218/0'
plc = LogixDriver(ipAndSlot, init_program_tags=True)
print(plc.tags)
```

AndrewStuhr, 2020-08-12:
As always, you're the best!

ottowayi, 2020-08-12:
Glad to help, have a good one

closed 2020-08-12 by ottowayi