ottowayi/pycomm3 #72: Read tags using their aliases

ceekay456, 2020-08-18:
Hi, 

I could not find it in the docs or the examples but is is possible to read/write tags based on the tags Alias?
The alias is not part of the Tag class I see...

ottowayi, 2020-08-18:
Yes, you can read a tag by either it's base or alias name.  If you're looking thru the tag list, you'll see that there is a _alias_  property to indicate if the tag is an alias or not.  But it might not even be accurate, I just found that one of the undocumented properties always had a bit True when it was an alias. So, I've been using that for now but I can't guarantee it'll always work since the is no mention of alias tags in the docs that I've been able to find.

ceekay456, 2020-08-18:
Ok clear. Thanks for the quick response.

closed 2020-08-18 by ceekay456