ottowayi/pycomm3 #74: Can't find array variables on Micro850

ceekay456, 2020-09-04:
hi, 

We're using a Micro850 PLC and when printing the tags, we notice that user defined array variables are not listed. 
They can also not be found when trying to read.

Simple user defined type variables like INT can be found.

See picture attached. Are we missing something in the code / settings? 
Our goal is to use pycomm3 to read/write arrays and structured variables.

Note: in the picture attached we're trying to read but as mentioned, using print(plc.tags) doesn't show the array variables either.

Thanks in advance,
Christophe

[image]

ottowayi, 2020-09-04:
It appears that it's related to the range of the array not beginning at 0.  I did a quick test with pylogix and it has the same issue as well. I'll look into it this morning and hopefully I can find something.

[image]
[image]
[image]

anandspande, 2020-09-04:
> It appears that it's related to the range of the array not beginning at 0. I did a quick test with pylogix and it has the same issue as well. I'll look into it this morning and hopefully I can find something.
> 
> [image]
> [image]
> [image]

Thanks a lot. starting index seems to be problem. 
i was going over documentation and i see below for arrays 
1. An array cannot be a Boolean data type.
2. An array cannot be an Array data type.
**3. The starting index of the array must be zero.**

ottowayi, 2020-09-04:
That's weird..  if it's not allowed then why are you letting me do it Rockwell?  I've been doing some research and it looks like most OPC servers don't support browsing for Micro800s.  And pycomm3 requires the tag list so that the structures reads/writes and automatic request size handling works.  Unfortunately, there is not an option to bypass that stuff right now.  The Micro800 support was added after all of the Logix stuff, in hindsight I probably should have made it a separate driver since they're much more limited than the Logix PLCs.  I'm sorry I don't have a solution to the problem right now, but I will think about it and see what I can come up with.  Either a separate driver for Micro800s or an option to bypass all the under-the-hood-magic and make it the user's responsibility to ensure the requests are accurate.  I'm about to head out on a startup though, so it may be a bit before I can get something together.

ottowayi, 2020-09-04:
And that reminds me, I need to fix those error messages for unknown tags, `'Invalid tag request - 'plc_tag'` is not very helpful

closed 2020-09-22 by ottowayi

cannonkalra, 2020-10-14:
@ceekay456 I'm also looking for the solution for this problem.

ottowayi, 2020-10-14:
I think the PLC simply doesn't include them in the tag list when you request it.  I looked at the raw data being returned and they aren't in there, so I don't think it's a parsing issue. I think it's because in the CIP spec, arrays only have a single dimension value. So rather than breaking things by not fitting the spec, they just leave them out. 

[image]