ottowayi/pycomm3 #75: Accessing String and Ascii files on Micrologix1400

saltratrix, 2020-09-04:
Great work here but I'm unclear on the correct syntax pycomm3 expects for accessing ST and A files. I am trying to make a simple reference to ST52:0 as show here. 
[image]
I'm cool stringing together bits from the A53 file as well.  
[image]

ottowayi, 2020-09-08:
It appears there never was support for string and ASCII files in pycomm.  I updated the SLC driver to use my base CIP driver for the for the actual EIP communications, but didn't change how it parses the requests and handles the different data file types.  I will look into adding those file types, but I can't promise when I will have it working.

saltratrix, 2020-09-08:
Thanks very much for the response, I needed it running in prod by this morning so  I hacked something together in C using libplctag; it's ugly as the string bytes are written inversely (for an 8 character string the data is in this order [1,0,3,2,5,4,7,6...]). I will close this for now thanks again.

closed 2020-09-08 by saltratrix

ottowayi, 2020-09-08:
Yeah I just started figuring that out, I think I have it working but I'm currently on a startup so not much time to really dig into it at the moment.

ottowayi, 2020-09-08:
@saltratrix I just pushed to develop some changes that should make the string and ascii files work.  I only have a SLC to test against, but it should be the same for MicroLogix.  I have a couple other things I want to do before releasing the next version, but if you want to test it out it's there.