ottowayi/pycomm3 #76: Does pycomm automatically close the connection?

AndrewStuhr, 2020-09-08:
In issue #32 , you mentioned calling plc.close() for a long lasting driver.  I haven't seen plc.close() in any example code.  Does pycomm3 automatically call plc.close under normal conditions, and under what circumstance should I manually call plc.close?

Thanks!

ottowayi, 2020-09-08:
If you use it in a context manager (`with LogixDriver ...`) then it will close it when you exit the block.  All the examples are simple ones in `with` blocks that why you don't see any calls to `close`.  If you're not using it in a context manager of some sort, then yeah you should call close.  The PLC will close the connection eventually, but I can't say for sure that it won't cause any issues over time.   In a couple of my apps I create a driver in a background thread, store it as an attribute, and have it read continuously once the user _connects_ to the PLC.  Then when they _disconnect_ and it shutdowns the thread, it calls `close`.

AndrewStuhr, 2020-09-08:
Thanks - that makes a lot of sense.  I've been using context managers throughout python but didn't know they called a close/exit method by design.  Now I know!

closed 2020-09-08 by AndrewStuhr

ottowayi, 2020-09-08:
Yep, they're pretty nifty. You can turn any class into one my adding `__enter__` and `__exit__` methods. If you look at the CIPDriver source you will see them in there.