ottowayi/pycomm3 #78: Cant read string using generic_message

Fryj2841, 2020-10-07:
Hey,

  I have been working on a project lately using generic CIP messaging to a Lexium mdrive stepper motor.  I have found an issue and am not sure where the problem is.  I have the following code.

`def get_setup_function(path):
    with CIPDriver(path) as driver:
        param = drive.generic_message(
            service=CommonService.get_attribute_single,
            class_code=b'\x67',
            instance=1,
            attribute=b'\x0F,
            connected=True,
            unconnected_send=False,
            route_path=False,
            data_format=[('IS','string')],
            name='data'
        )
    return param[1]`

I dont get any errors just a return of None.  I have tried to set the data_format to None and I get a bytestring but the data isnt complete. Any help would be great

Thanks

ottowayi, 2020-10-07:
Try setting the data type to `'SHORT_STRING'` in the data format, I believe that is the CIP data type most attributes use.  You can find the different data types [here](https://pycomm3.readthedocs.io/en/latest/cip_constants.html).  What value are you expecting from it?  Also, if you can share that raw data that might help.  If you set the format to None, it returns the raw data so it can be parsed externally if it's not supported directly.

Fryj2841, 2020-10-07:
Thanks for the quick response.  Sorry for the code format.  I typed it in a lot nicer than that but this is my first time messaging someone on GitHub so I didn't know how to format it.  I have already tried to use the 'SHORT_STRING' data type as well and still get a response of None.  The response I am expecting is a multi-line text response showing several lines of text for the setup of the drive I am working on.  The raw data I am receiving comes back as a single line.  

  b'\x0e\x00\x00\x00IS = 1, 0, 1\r\n'

I am expecting a first line of IS = 1, 0, 1 and I see that it is coming in correctly for the first line but I need to see 5 different lines of this string data so it would look like this inside of a terminal window.

 IS = #, #, # 
     = #, #, # 
     = #, #, # 
     = #, #, # 
     = #, #, #

Thanks again for the help. 

Regards,

ottowayi, 2020-10-08:
So I looked at the documentation for that drive, and  you're correct the attribute is a STRING type.  But, I noticed that I only account for SHORT_STRING types.  I just pushed a couple changes to the develop branch that should fix that.  If you're not familiar with installing from github, you can use this pip command to get it: `pip install --upgrade git+https://github.com/ottowayi/pycomm3@develop`.  I didn't test it though, so let me know how it works. I can also show you where those changes are made if installing from gitub doesn't work. 

[image]

This change also doesn't fix the same issue for STRING2 or STRINGN types, if they're needed I can add them at some point too.

Fryj2841, 2020-10-09:
Thank you so much.  I will try this tomorrow and let you know how it works.

ottowayi, 2020-10-09:
No problem, let me know if it works and I'll publish a new version.

Fryj2841, 2020-10-09:
So I tested this out and I can get a string out of the data using the the following code. 

[image]

This code shows this response.

[image]

but when I try to get the data from the dictionary by using data=param[1]['IS'] we get a blank response.

ottowayi, 2020-10-09:
I'm not sure what issue could be, it looks correct.  `param` is a named tuple, so you could use `param.value` instead of `param[1]` and both of those should be giving you the same dict back.  

Can you try adding these and share what the result is?
```

print('IS' in param.value)
print(param.value)
print(repr(param))

```

ekkizogloy, 2020-10-12:
It looks like @Fryj2841 beat me to it. My project involves accessing a different PLC and I have a need to decode a string type (DataType = string =0xd0). I would prefer to just use pycomm3 as is and not fork it. This decode is the same as a the short_string, except that the string length is 2 bytes long. I added the functions to bytes_.py and would like to submit a PR. What process do you prefer? Let me know and I'll send it over.

ottowayi, 2020-10-12:
Normally, I would say just send a pull request and I'll most likely merge it as long as it doesn't break anything.  But, first can you try pulling from the develop branch and see if that works?  I added those functions, but didn't test them against a PLC.  If those don't work and yours do, let me know and send the PR.  Else, I'll just merge develop and make a new release.

Fryj2841, 2020-10-12:
Hey,

  This is the result from the triple print statement.  

[image]

  It is definatly retruning a dictionary object because I have asked it to return the type on value and it responds with <ckass 'dict'>  but I still cannot get the value from the dictionary by using ['IS'].  It just returns a blank line.  Also I am still only getting the first line of text.

Regards,

ottowayi, 2020-10-12:
I wonder if it's a display issue in whatever you're using to view the value since the string starts with 2 nulls.  Can you try calling `strip` on the string before printing it?  If that value wasn't in there you'd be getting a `KeyError` when trying to do `param.value['IS']`.  As for it only returning one-line, is it possible you need to make multiple requests to get all the data?  As a `STRING` type, it stores the length of the string in the first 2 bytes.  So what you're seeing should be all that is there for the data for that tag. Maybe you'll need to use different requests or something to get the remaining data. Are you able to use a MSG instruction in Logix to get the full data?  Would you be able to remove the `data_format` arg and share the full byte string it returns?

ekkizogloy, 2020-10-12:
> Normally, I would say just send a pull request and I'll most likely merge it as long as it doesn't break anything. But, first can you try pulling from the develop branch and see if that works? I [...]

Thanks @ottowayi . It looks like the `develop` branch works! Thanks so much.

ottowayi, 2020-10-12:
@ekkizogloy glad it works, I'll publish a new release sometime (hopefully) soon. So if you're watching the repo you'll get a notification.

Fryj2841, 2020-10-12:
Thanks again for all the help. Stripping the nulls fixed the blank line issue.  I am able to remove the data_format arg and I am getting a byte string as follows.

[image]

If I am not mistaken if the first couple of bytes are the length of the string we should have a string with a length of 14 which corresponds to only the first line of data.  I will keep playing with this to try and get all of the corresponding lines of data necessary.  

Regards,

Fryj2841, 2020-10-12:
So the actual message that should be coming through should be 6 lines of data all in the same format as the one above.  It looks like this in the window that Lexium provide

[image]

So I'm a little confused as to why we are not getting the length of the whole string in the first read.  Should I contact the drive manufacturer?

Thanks

ottowayi, 2020-10-12:
Yeah I'm not sure what the issue may be.  You could try using a MSG instruction in the PLC and have it write to a custom UDT for the string with an INT for the len, followed by a SINT[100] data array and see if you get more than 14 characters out of it.  Or reaching out to them might solve it too.  They probably won't be too helpful if you saying you're using this library, so I would lead with you're trying to get it working with a PLC message first. The other option is to do a wireshark capture when you're using that program and we can try and view the requests it's making.

Fryj2841, 2020-10-12:
Wireshark could be a great solution I am also not going through the PLC I am accessing the device straight from your library.  I am trying to avoid connecting it to a PLC.

ottowayi, 2020-10-18:
string fixes added in 0.11.0
I also added a `logix_string` option for Pack/Unpack, since in your case that tag is actually a Logix string tag, which has 4-bytes for the length of the string.  The CIP string type only uses 2 bytes for the length, so that's why you had those 2 null characters at the start of the string.  If you change the data format to `'logix_string'` it should fix that.

closed 2020-10-18 by ottowayi