ottowayi/pycomm3 #79: [FEATURE] - type hints or library stubs for library users

EduardoHi, 2020-10-10:
First of all thanks for this useful library !
Now, I noticed that pycomm3 uses typing hints and typechecking internally, But when using `mypy` I get the error:
`Skipping analyzing 'pycomm3': found module but no type hints or library stubs` (I might be doing something wrong with my setup so I apologize if that's the case and would appreciate guidance on how to set that up properly).

Here's more information about it 
from the users perspective: https://mypy.readthedocs.io/en/stable/running_mypy.html?#missing-type-hints-for-third-party-library
and from the maintainers perspective: https://www.python.org/dev/peps/pep-0561/

**Type of Feature Request**
- [x] missing feature
  - [ ] feature available in another library: _please specify_
- [ ] change to API
- [x] enhancing a current feature
- [ ] removal of a broken/unsupported/etc feature
- [ ] other: _please specify_

**Feature Description**
A way to have library users typecheck their usage of pycomm3.

**Possible Solution**
From what I understand there are 3 different approach that might apply here:
1. Add `py.typed` marker file to signal mypy that type information is inlined.
2. Add `*.pyi` stubs files
3. create a separate `pycomm3-stubs` package and let users optionally dowload that

With some guidance I am willing to do the work and send PR, let me know what you think !

ottowayi, 2020-10-11:
Hey I think this is a good idea and would definitely merge a PR if you send one.  I think option 1 seems like the easiest solution.  I haven't used mypy, so if it only needs a single file it seems like the simplest route.  I'm not a big fan of having to maintain something extra since the type hints are already in the code, but if it's required I'd definitely prefer option 2 over 3. I've tried to add them to all the public interfaces, but I still need to add them internally too.  So yeah, if you get it working and send a PR I will accept it.

ottowayi, 2020-10-18:
added in version 0.11.0

closed 2020-10-18 by ottowayi