ottowayi/pycomm3 #80: List(s) of tags

ccscpanici, 2020-10-12:
I need to give the read function a list of tags or convert that to something the read function can understand - I tried a tuple but that didn't work. 
I need to give the write function a list of tags and values or convert that to something the convert that to something it would understand.
For example:

tags = ['tag1', 'tag2', 'tag3']
results = plc.read(tags)

or something like.....
tag_q = CType(<something>, tags)
results = plc.read(tag_q)

Any information you can provide to help me would be greatly appreciated. I thought before I go forking the code and overriding these methods I would ask you.

CJ

ccscpanici, 2020-10-13:
Update: I just remembered about the * in front of lists will unpack them. 
plc.read(*tags) works.

CJ

closed 2020-10-13 by ccscpanici

ottowayi, 2020-10-13:
Yeah, that's the correct way to do it. I chose that way because I felt it was simpler without having to check if a tag was a list/tuple/etc and really multiple tags.   Using a `*` arg seemed like the best way.