ottowayi/pycomm3 #85: Get PLC clock metadata like timezone, daylights saving adjustments?

ambersnow, 2020-11-02:
Hi!

I know you can get/set PLC time, which we are already using. Is there a way we can check what timezone the PLC was using, if it is adjusting for daylights saving, etc. Like how it is shown here:
[image]

Thanks in advance!

ottowayi, 2020-11-02:
Yeah it's possible, just not built in.  Using [this](https://literature.rockwellautomation.com/idc/groups/literature/documents/qs/2080-qs002_-en-e.pdf) doc, we can get the attributes for the Wall-Clock object and then use a generic message to read them.  For some reason, I can't get the `get_single_attribute` to work and just keep getting an error that it isn't supported, but using a `get_attribute_list` service does.  But, there is some extra data returned when using the attribute list service, so I account for that in the data format. 

Try something like this:
```
print(plc.generic_message(
    service=CommonService.get_attribute_list,
    class_code=ClassCode.wall_clock_time,
    instance=0x01,
    request_data=b'\x02\x00\x0A\x00\x08\x00',  # 2 attributes, #10 and #8
    data_format=[
             (None, 2),  # UINT - number of attributes
             (None, 4),  # UINT - attr id, UINT - attr status
             ('dst', 'USINT'),  # 1 - DST enabled, 0 - disabled
             (None, 4),  # UINT - attr id, UINT - attr status
             ('tz', 'logix_string'),  # time zone string
    ],
))
```

and it prints:

`Tag(tag='generic', value={'dst': 1, 'tz': 'UTC-06:00 Central Time (US & Canada)'}, type=None, error=None)`

which matches my PLC:
[image]

ambersnow, 2020-11-02:
Awesome thank you! I will give this a shot.

closed 2020-11-02 by ambersnow