ottowayi/pycomm3 #86: None of the documentation describes how to interact with binary addresses (Micrologix)

lionenergyengineering, 2020-12-02:
I'm trying to write to a binary address, bit specific. For example, B3:0/0. I am able to write a value to it at the word level, meaning I can change B3:0 to a 1, or an 8, or 32. But again, I am unable to write a 0 or a 1 to B3:0/0.

Surely it must be possible?

ottowayi, 2020-12-02:
The SLCDriver only has minor changes made to it to make it fit into the API changes for pycomm3, but the majority of the code remains the same.  I originally had to plans of migrating it, but due to the number of requests I ported it to Python 3.  I also don't have the hardware to test it, so any issues or fixes need to be tested by users.  If you are willing to identify or fix issues, I will accept a pull request or offer suggestions.  There isn't any documentation written for it because it wasn't the main goal of what I was trying to achieve, so it doesn't get high priority when I have time to work on pycomm3. 

Can you share the errors you are receiving?  It's possible something was broken when porting the code.
Have you tried writing to bit addresses for other types (maybe N files)?  If that is working, then it's something specific to B files.

ottowayi, 2020-12-02:
Actually, I may have found something.  

In the pycomm3/slc.py module there is a function called `writeable_value`.  Around line 626, there is a few function calls to `math.pow(2, bit_position)`.  `math.pow()` returns a float, which would throw an exception when passed into the Pack methods that expect and int.  Can you try changing those 3 instances to be `Pack.uint(2**bit_position)` instead of using `math.pow`?

lionenergyengineering, 2020-12-02:
This works:
`from pycomm3 import SLCDriver

with SLCDriver('192.168.1.80') as plc:
    plc.write(('B3:0', 1))
    plc.write(('B3:0', 32))`

The above writes at the word level. Writing to an Integer file works as well.

This does not work:
`from pycomm3 import SLCDriver

with SLCDriver('192.168.1.80') as plc:
    plc.write(('B3:0/0', 1))`

Here is the error
> pycomm3.exceptions.RequestError: Failed to create a writeable value for B3:0/0 from 1

And here is the full error

> C:\Users\j\PycharmProjects\sdfsdlafjlskad\venv\Scripts\python.exe C:/Users/j/PycharmProjects/sdfsdlafjlskad/main.py
Traceback (most recent call last):
  File "C:\Users\j\PycharmProjects\sdfsdlafjlskad\venv\lib\site-packages\pycomm3\slc.py", line 627, in writeable_value
    _value = Pack.uint(math.pow(2, bit_position)) + Pack.uint(math.pow(2, bit_position))
  File "C:\Users\j\PycharmProjects\sdfsdlafjlskad\venv\lib\site-packages\pycomm3\bytes_.py", line 122, in <lambda>
    uint: Callable[[int], bytes] = lambda n: pack('<H', n)
struct.error: required argument is not an integer

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:/Users/j/PycharmProjects/sdfsdlafjlskad/main.py", line 4, in <module>
    plc.write(('B3:0/0', 1))
  File "C:\Users\j\PycharmProjects\sdfsdlafjlskad\venv\lib\site-packages\pycomm3\cip_base.py", line 62, in wrapped
    return func(self, *args, **kwargs)
  File "C:\Users\j\PycharmProjects\sdfsdlafjlskad\venv\lib\site-packages\pycomm3\slc.py", line 135, in write
    results = [self._write_tag(tag, value) for tag, value in address_values]
  File "C:\Users\j\PycharmProjects\sdfsdlafjlskad\venv\lib\site-packages\pycomm3\slc.py", line 135, in <listcomp>
    results = [self._write_tag(tag, value) for tag, value in address_values]
  File "C:\Users\j\PycharmProjects\sdfsdlafjlskad\venv\lib\site-packages\pycomm3\slc.py", line 169, in _write_tag
    writeable_value(_tag, value),
  File "C:\Users\j\PycharmProjects\sdfsdlafjlskad\venv\lib\site-packages\pycomm3\slc.py", line 635, in writeable_value
    raise RequestError(f'Failed to create a writeable value for {tag["tag"]} from {value}') from err
pycomm3.exceptions.RequestError: Failed to create a writeable value for B3:0/0 from 1

lionenergyengineering, 2020-12-02:
> Actually, I may have found something.
>
> In the pycomm3/slc.py module there is a function called `writeable_value`. Around line 626, there is a few function calls to `math.pow(2, bit_position)`. [...]

I will try giving that a shot

lionenergyengineering, 2020-12-02:
I changed those three instances like you said, and that does not do anything. HOWEVER, I do not get an error anymore. The script finishes with exit code 0.

This is what I changed it to
```
if value > 0:
    _value = Pack.uint(2**bit_position) + Pack.uint(2**bit_position)
else:
    _value = Pack.uint(2**bit_position) + Pack.uint(0)
```

ottowayi, 2020-12-02:
Hmm.. I don't see any examples from pycomm of them using bit level writes, but I wonder if it did work and I broke it.  It looks like I changed the write command from 0xAB to 0xAA, doing some more googling 0xAB is an undocumented write with mask function.  I may have changed it to 0xAA because that is what was documented, either intentionally or accidentally - I don't remember doing it.  Can you try changing it in const.py to `SLC_FNC_WRITE = b'\xab'` and see if that works?  I'm not sure if that is all that needs to be done, I'll have to dig in more and see if it there were any other changes.  I don't have anything to test against at the moment, so hopefully we can get this sorted out quickly.

lionenergyengineering, 2020-12-03:
I can confirm that pycomm did work at the bit level. And awesome! I’ll make these changes tomorrow and I’ll get back to you

lionenergyengineering, 2020-12-03:
I am happy to report that after changing `SLC_FNC_WRITE = b'\xaa'` to `SLC_FNC_WRITE = b'\xab'` that it now writes at the bit level :) I haven't done any other testing though, so I have no idea if we broke anything else by making this change. And I don't know if it would be possible anyway. To be honest, this "lower level" python programming is not something I ever do, so I'm definitely out of my element here. But so far so good :)

ottowayi, 2020-12-03:
Awesome, I'm glad that worked.  I just pushed the fix to the develop branch.  If you are able to pull from the develop branch and test it out, I can release an update.  I don't have access to a SLC or MLX right now (and probably won't for awhile), so I'll have to rely on someone else to test it for me.  I just don't want to release a broken version.  If you can't test it, I have a couple others that I know are using it and can ask them.

lionenergyengineering, 2020-12-03:
Sounds good! I can't test it right this second, but possibly this weekend I'll be able to

ottowayi, 2021-01-16:
This should be fixed in 0.12.0

closed 2021-01-16 by ottowayi