ottowayi/pycomm3 #87: Writing a structure

JoeRecursionJoe, 2020-12-08:
I have some user defined types I need to write to.

The UDT is structured as follows:

**UDT Name**: PR_Real
- **Description** - _STRING_
- **Min** - _REAL_
- **Max** - _REAL_
- **Units** - _STRING_
- **AccessLevel** - _DINT_

TA_0EB01 is of type PR_Real
Doing a test, this works:
```
from pycomm3 import LogixDriver
PRB05_PLC = LogixDriver('10.9.1.187')

PRB05_PLC.write(('TA_0EB01.Description', 'BDC Cam'))
PRB05_PLC.write(('TA_0EB01.Min', 10.0))
PRB05_PLC.write(('TA_0EB01.Max', 80.0))
PRB05_PLC.write(('TA_0EB01.Value', 5.6))
PRB05_PLC.write(('TA_0EB01.AccessLevel', 9999))
```

Is there a way to use a dictionary or list like this without writing a loop?
```
PRB05_PLC = LogixDriver('10.9.1.187')
tagvalues = {'Description': 'BDC Cam', 'Min': 10.0, 'Max': 80.0, 'Value': 5.6, 'AccessLevel': 9999}
PRB05_PLC.write(('TA_0EB01', tagvalues))
```

The example writes for UDT shows using several lists, but it was unclear how the recipe example was structured in the PLC?

ottowayi, 2020-12-08:
So the structure writing using sequences of values (list, tuple, etc) instead of a dictionary.  So for your use case, if you get rid of the keys and make `tagvalues` a list of just the values, then it should work.  You would then nest lists for arrays or other structs.  I will update the example with comments to show how the values are mapped to the struct. 

I think adding support for dicts is a good idea.  I can look at adding it as a feature.

JoeRecursionJoe, 2020-12-08:
When I try a list, I get an error, also, I'm not sure how it knows which value goes where.

If I do a read I get the following:
```
>>> print(PRB05_PLC.read('TA_0EB01'))
TA_0EB01, {'AccessLevel': 9999, 'Description': 'BDC Cam', 'Max': 200.0, 'Min': 10.0, ...}, PR_Real, None
```

If I try a write and keep the same order I get an error
```
>>> tagvalues = [9999, 'Test', 80.0, 10.0, 5.6]
>>> PRB05_PLC.write(('TA_0EB01', tagvalues))
```

```
Traceback (most recent call last):
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 1204, in _pack_structure
    value_bytes = [pack_func(val), ]
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\bytes_.py", line 130, in <lambda>     
    real: Callable[[float], bytes] = lambda r: pack('<f', r)
struct.error: required argument is not a float

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 1129, in writable_value
    return _writable_value_structure(value, elements, data_type)
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 1157, in _writable_value_structure
    return _pack_structure(value, data_type)
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 1210, in _pack_structure
    raise RequestError('Value Invalid for Structure') from err
pycomm3.exceptions.RequestError: Value Invalid for Structure

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "f:\Programming\Source Control\Python\SetToolData.py", line 13, in <module>
    PRB05_PLC.write(('TA_0EB01', tagvalues))
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\cip_base.py", line 62, in wrapped
    return func(self, *args, **kwargs)
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 783, in write
    requests, bit_writes = self._write_build_requests(parsed_requests)
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 815, in _write_build_requests
    return [r for r in requests if r is not None], bit_writes
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 815, in <listcomp>
    return [r for r in requests if r is not None], bit_writes
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 814, in <genexpr>
    requests = (self._write_build_single_request(parsed_tags[tag], bit_writes) for tag in parsed_tags)
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 869, in _write_build_single_request
    parsed_tag['write_value'] = writable_value(parsed_tag)
  File "C:\Users\joer.PR\AppData\Local\Programs\Python\Python39\lib\site-packages\pycomm3\clx.py", line 1138, in writable_value
    raise RequestError('Unable to create a writable value') from err
pycomm3.exceptions.RequestError: Unable to create a writable value
PS F:\Programming\Source Control\Python>
```

ottowayi, 2020-12-08:
Okay I think I see what's happening, when you print out the `Tag` object it uses reprlib to truncate really long values when calling the __str__ method.  If you do `print(PRB05_PLC.read('TA_0EB01').value)` you will see the full structure and values.  Then when you go to write it, the value list must match the same order as the attributes are defined in the PLC.   So for your UDT values would need to be [Description, Min, Max, Units, AccessLevel].  The value dict should be in order since you're on 3.9 which has ordered dicts by default, but I wonder if reprlib is messing with it when it converts them to a string.  I'll have to look into that, but in general until I added the dict-support to the write method you'll have to have your values in the order of the attributes.

JoeRecursionJoe, 2020-12-08:
Thanks, list option works now.
I had also forgotten to include the "Units" string.

```
>> print(PRB05_PLC.read('TA_0EB01').value)
{'Value': 56.400001525878906, 'Min': 10.0, 'Max': 200.0, 'Units': 'mm', 'Description': 'Test', 'AccessLevel': 9999}

>> tagvalues = [12, 9.3, 100.4, 'mm', 'Test list']
>> PRB05_PLC.write(('TA_0EB01', tagvalues))
{'Value': 12.0, 'Min': 9.300000190734863, 'Max': 100.4000015258789, 'Units': 'mm', 'Description': 'Test list', 'AccessLevel': 0}
```

When you make a Dict option, can you have it to where you do not have to include every subparameter for example:
{'Min': -1.0, 'Max': 250.0, 'Units': 'mm'}
or
{'Value': 12.0, 'Description': 'Stopping Cam'}
or
{'Value': 12.0, 'Min': 9.300000190734863, 'Max': 100.4000015258789, 'Units': 'mm', 'Description': 'Stopping Cam', 'AccessLevel': 0}

All of these would work. the values not keyed would remain unchanged.

ottowayi, 2020-12-08:
That would be desirable, but it introduces complexity of how to actually achieve it.  When writing a struct it operates like the COP instruction, where it's a full binary copy of data into the struct, so if we want omitted values to remain unchanged the same values would have to be written back to it.  So should the write method call a read first if all the data is not supplied?  Or should the user do a read and merge the two dicts before calling write? And how should it handle recursively checking any nested structs?  I think having it all handled in the write method would be the simplest for the user personally.  

Also, I looked at the reprlib source and it is calling `sorted` on the dict before converting it to a string.  I've found a workaround and will get it fixed soon.

JoeRecursionJoe, 2020-12-08:
Instead of writing it as a struct, maybe if it is a dict just iterate through each value and write them as single values.
This may not be as efficient, I don't know, but could be an option.

for example:

user.py
```
tagvalues = {'Value': 101, 'Min': 10.0, 'Max': 200.0}
PRB05_PLC.write(('TA_0EB01', tagvalues))
```

pseudo code in clx.py
```
if type(x) == type(dict()):
    for key, value in x.items():
        PLC.write((f'{basetag}.{key}', {value}))
```

ottowayi, 2020-12-08:
That's possible too, what I was thinking something like this:

```
# in the write method
if isinstance(value, Mapping):
    if missing_data(value):   # method to compare value dict to the data type definition
        value = merge_values(value, self.read(tag).value)   # fill in any missing values with current values
    writeable_value = writeable_from_dict(value)
else:
    writeable_value = writable_from_list(value)  # these could easily be the same method
```

JoeRecursionJoe, 2020-12-08:
The only thing I wonder is if a value you're not writing to changes between the time you read it and the time you write it, you would overwrite the changed value, however that may not be an issue....

ottowayi, 2021-01-15:
Added in v0.12.0

closed 2021-01-15 by ottowayi