ottowayi/pycomm3 #93: I think this project should have Contribution Guidelines

zenware, 2021-01-16:
This basically amounts to a `contributing.md` file in the root of the project which explains the expectations of contributors to the project. Typically it will include things like "what are the expectations of an Issue, a PR.. what code/comment styleguide is meant to be followed. And any expectations on the git workflow itself, i.e. is work done in feature branches, is work done on a development branch, etc. It may also include a code of conduct and any other information deemed useful. It could also bs very simple and brief, even to the point where all of it fits inside the main readme file, as some projects do.

I bring this up because in my late night contribution I didn't notice there was a 'develop' branch, which is as of this writing 1 commit ahead, 2 commits behind master. I now sort of expect that I should've requested my PR be merged into develop rather than master.

Here's an example of my favorite short and sweet contributing file https://github.com/shazow/ssh-chat/blob/master/CONTRIBUTING.md

Github has a document with a guide for this that provides links to several examples near the bottom. 

https://docs.github.com/en/github/building-a-strong-community/setting-guidelines-for-repository-contributors

zenware, 2021-01-16:
Apologies for using the question issue type for what is arguably a feature request. I was unsure whether issues or the discussion page were the right platform for this and I'm more familiar with issues so I went that route.

ottowayi, 2021-01-16:
Yeah that is definitely something I need to get added, I'll work on putting one together.  I merged your PR to master because it was small and unlikely to cause conflicts and I have some code in develop that I'm still working on.  I really should have done it in a feature branch.  As for issues vs discussions... I just enabled the discussions because I saw it was new.  I suppose questions could go in either, so I don't have a strong opinion either way.

ottowayi, 2021-01-18:
added

closed 2021-01-18 by ottowayi