ottowayi/pycomm3 #96: Trying to write to an element in a Data File

xinthose, 2021-01-26:
Hello.  I would simply like to write a 1 into Data File N11:1.  Am I missing something in my code?

```
            with LogixDriver(plc_ip_address) as plc:
                plc.write(("N11[1]", 1))
```

**Log output:**

```
2021-01-26 08:38:54,847 [DEBUG]: porter_rest.py(GET:74) >> plc_ip_address = 10.0.0.5
2021-01-26 08:38:54,862 [DEBUG]: requests.py(send:133) >> Sent: RegisterSessionRequestPacket(message=[b'\x01\x00', b'\x00\x00'])
2021-01-26 08:38:54,871 [DEBUG]: requests.py(send:139) >> Received: RegisterSessionResponsePacket(session=930230157, error=None)
2021-01-26 08:38:54,871 [INFO]: cip_base.py(_register_session:252) >> Session = 930230157 has been registered.
2021-01-26 08:38:54,872 [DEBUG]: requests.py(send:133) >> Sent: ListIdentityRequestPacket(message=[])
2021-01-26 08:38:54,881 [DEBUG]: requests.py(send:139) >> Received: ListIdentityResponsePacket(identity={'item_type_code': 53, 'item_length': 1, 'encap_protocol_version': 512, '_socket_address_struct': b'\xaf\x12\n\x00\x00\x05\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00', 'vendor_id': 14, 'product_code': 90, 'revision_major': 2, 'revision_minor': 11, 'status': 4, 'serial_number': 1080144404, 'product_name': '1766-L32BWA B/11.00', 'state': 0}, error=None)
2021-01-26 08:38:54,881 [DEBUG]: requests.py(send:133) >> Sent: GenericUnconnectedRequestPacket(message=[b'R\x02 \x06$...1\x00\x01\x00'])
2021-01-26 08:38:54,891 [DEBUG]: requests.py(send:139) >> Received: GenericUnconnectedResponsePacket(service=b'R', command=b'o\x00', error='Service not supported - No Extended Status')
2021-01-26 08:38:54,891 [ERROR]: porter_rest.py(GET:85) >> Failed to get PLC info
```

**My PLC setup:**

[image]

ottowayi, 2021-01-26:
`N11:1` should be the correct format, but you're connecting to a MicroLogix which means the LogixDriver won't work.  The naming is sort of confusing, but the LogixDriver is only for the _modern_ controllers (ControlLogix, CompactLogix, Micro800).  The MicroLogix needs to use the SLCDriver since it uses the old protocols and doesn't support the same features.  Where your code is failing is the driver is trying to upload some information about the PLC that just doesn't work on the old stuff.  Try the SLCDriver and see if that works.

xinthose, 2021-01-26:
@ottowayi that worked!  Thank you.  My code for completeness:

```
    with SLCDriver(plc_ip_address) as plc:
        plc.write(("N11:1", 1))
```

I missed this [note](https://pycomm3.readthedocs.io/en/latest/slcdriver.html) on your website, sorry:

> An Ethernet/IP Client driver for reading and writing of data files in SLC or MicroLogix PLCs.

closed 2021-01-26 by xinthose

ottowayi, 2021-01-26:
Glad that worked.  The 'Logix' part of MicroLogix is sort of confusing on which driver to use. Originally I never planned on migrating the SLC driver from pycomm to pycomm3, but it was a common enough request that I attempted it.  It's still pretty much the same as it was in pycomm, but I changed the API to match the LogixDriver (single read/write methods, return Tag objects, etc).  I haven't got around to any documentation or tests for it yet, but if there are bugs I can try and fix them.

xinthose, 2021-01-26:
I see.  We buy hundreds of MicroLogix 1400's every year where I work, so they are still being used.