ottowayi/pycomm3 #98: Tag Listing

GitHubDragonFly, 2021-01-31:
Hi there.

I am not much of a python programmer and am looking for some advice and/or input on how to simplify the code for tag listing within listbox of the tkinter GUI.

The output of the code looks like in the attached picture.

[image]

I did try using pylogix but it currently doesn't provide UDT members info.
The help I am looking for is how to make the process recurring to penetrate "n" levels of UDT and hopefully eliminate the struct_members2 function (which is almost the same as struct_members function) from the following code:

`def getTags():
    lbTags.delete(0, 'end') #clear the tags listbox

    comm_check() #check if either of IP Address / Processor Slot was changed

    tags = comm.get_tag_list('*') #get all tags

    currentTagLine.set(1) #start at the first line of the tags listbox

    if not tags is None:
        j = 1

        for tag, _def in comm.tags.items():
            #-----------------------------------------------------------------------

            dimensions = ""
            dim = _def['dim']

            if dim != 0:
                dims = str(_def['dimensions'])[1:-1].split(',')

                if dim == 1:
                    dimensions = "[" + dims[0] + "]"
                elif dim == 2:
                    dimensions = "[" + dims[0] + "," + dims[1] + "]"
                else:
                    dimensions = "[" + dims[0] + "," + dims[1] + "," + dims[2] + "]"

            #-----------------------------------------------------------------------

            if _def['tag_type'] == 'struct':
                structureDataType = _def['data_type']['name']
                structureSize = _def['data_type']['template']['structure_size']

                lbTags.insert(currentTagLine.get(), tag + dimensions + " (" + structureDataType + ")" + " (" + str(structureSize) + " bytes)")
                currentTagLine.set(currentTagLine.get() + 1)

                response = struct_members(_def['data_type']['internal_tags'], currentTagLine.get(), j)

                while not response is None:
                    j += 1
                    response = struct_members(response, currentTagLine.get(), j)
            else:
                lbTags.insert(currentTagLine.get(), tag + dimensions + " (" + _def['data_type'] + ")")

            #-----------------------------------------------------------------------

            j = 1
            currentTagLine.set(currentTagLine.get() + 1)
    else:
        lbTags.insert(currentTagLine.get(), 'No Tags Retrieved')

def struct_members(it, i, j):
    # internal tags keys
    keys = it.keys()

    for key in keys:
        tag = it[key]

        if tag['tag_type'] == 'struct':
            structureDataType = tag['data_type']['name']
            structureSize = tag['data_type']['template']['structure_size']

            add_Tag(j, "- " + key + " (" + structureDataType + ")" + " (" + str(structureSize) + " bytes)")

            currentTagLine.set(i + 1)

            struct_members2(tag['data_type']['internal_tags'], currentTagLine.get(), j + 1)
        else:
            if tag['data_type'] == "BOOL":
                add_Tag(j, "- " + key + " (offset " + str(tag['offset']) + ")" + " (bit " + str(tag['bit']) + ")" + " (" + tag['data_type'] + ")")
            else:
                if tag['array'] > 0:
                    add_Tag(j, "- " + key + "[" + str(tag['array']) + "]" + " (offset " + str(tag['offset']) + ")" + " (" + tag['data_type'] + ")")
                else:
                    add_Tag(j, "- " + key + " (offset " + str(tag['offset']) + ")" + " (" + tag['data_type'] + ")")

            currentTagLine.set(currentTagLine.get() + 1)

    return None

def struct_members2(it, i, j):
    # internal tags keys
    keys = it.keys()

    for key in keys:
        tag = it[key]

        if tag['tag_type'] == 'struct':
            structureDataType = tag['data_type']['name']
            structureSize = tag['data_type']['template']['structure_size']

            add_Tag(j, "- " + key + " (" + structureDataType + ")" + " (" + str(structureSize) + " bytes)")

            currentTagLine.set(i + 1)

            return struct_members2(tag['data_type']['internal_tags'], currentTagLine.get(), j + 1)
        else:
            if tag['data_type'] == "BOOL":
                add_Tag(j, "- " + key + " (offset " + str(tag['offset']) + ")" + " (bit " + str(tag['bit']) + ")" + " (" + tag['data_type'] + ")")
            else:
                if tag['array'] > 0:
                    add_Tag(j, "- " + key + "[" + str(tag['array']) + "]" + " (offset " + str(tag['offset']) + ")" + " (" + tag['data_type'] + ")")
                else:
                    add_Tag(j, "- " + key + " (offset " + str(tag['offset']) + ")" + " (" + tag['data_type'] + ")")

            currentTagLine.set(currentTagLine.get() + 1)

    return None

def add_Tag(j, string):
    #insert multiple of 2 spaces, depending on the depth, to simulate tree appearance
    k = 2 * j + len(string)
    lbTags.insert(currentTagLine.get(), (" " * k + string)[-k:])
`

The full GUI picture looks like the attached, which was initially created for pylogix.

[image]

GitHubDragonFly, 2021-01-31:
I have managed to simplify the code and have edited my initial post to remove the old code and replace the py file with the currently updated one. Besides for pycomm3 it still requires pylogix for Device Discovery.

closed 2021-01-31 by GitHubDragonFly

coltman151, 2021-01-31:
Device discovery will be pushed to the master branch soon. See #91.

GitHubDragonFly, 2021-02-01:
When that happens then I can go about changing it.

In the meantime, I did apply a few more updates to the GUI and replaced the files in my initial post.

ottowayi, 2021-02-01:
I would be willing to include this in the examples if you want it shared.  If you do, please read the contributing instructions and create a new pull request.  Examples should be in the _examples_ directory.   Please include some comments or docstrings to explain why and how the code works, I will use that to write up a page in the docs and then include the example.

GitHubDragonFly, 2021-02-01:
It is still using pylogix for the device discovery.

Once you push your version then I can upgrade the pycomm3 installation on my computer, remove pylogix, modify the GUI and submit it in a PR.

ottowayi, 2021-02-01:
There is minor change that could be breaking for some users currently in the develop branch.   I'm working on a pretty big internal re-write right now that I'm going to finish before releasing a new version, that way any breaking changes are limited to a single release.  If you want to test out your code with the discover functionality, you can install from the develop branch.  The only 'breaking' change is that `open` is not called automatically during the init of the driver.  It doesn't affect using it inside a `with` statement, but if you're doing something like `plc = LogixDriver(...)` just know that you have to call `plc.open()` before using it.

GitHubDragonFly, 2021-02-02:
I have tried your device discovery and the string that shows for MicroLogix 1100 has these entries (amongst the others):

1) 'item_length': 1
2) 'vendor_id': 12
3) '_socket_address_struct': b'\xaf\x12\xc0\xa8\x01\n\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00'

The 1st entry appears to be the actual Vendor ID, the 2nd entry shows in pylogix as Device ID and the 3rd entry has "n" in the string (which should be "a" or "xa0").

Can you let me know if these are (in)correct?

ottowayi, 2021-02-02:
`discover()` returns a list of dictionaries, each dictionary is the Identity Object of the device.  You'd probably be most interested in the `'product_code'`, `'product_name'`, and `'ip_address'` fields.   If you want more information, look at the CIP Vol 1 doc in the `specs` folder.

GitHubDragonFly, 2021-02-02:
The PR has been created and files in my initial post either updated or removed.

GitHubDragonFly, 2021-02-05:
I will put the following info here just in case if anybody might be looking for such things:

Pydroid 3 app on Android devices can run this project with your library.
GUI file has to be opened in the IDE and run from there (terminal doesn't support running GUI).
Depending on the Android device and its resolution, one might need to make size and placement adjustments to widgets used in the GUI file.

There also seems to be another way of making things work and not only on Android:

dmroeder/pylogix#137

GitHubDragonFly, 2021-02-13:
Can you let me know if you do not want to accept the latest pull request?

This so I can close the PR and post that file as it is in my own repo.

ottowayi, 2021-02-13:
Sorry I haven't had the time I thought I would recently to finish the work I've been doing.  I merged it to my develop branch since it relies on the discover method that isn't in master yet. Hopefully I can get a new release out soon.  

How would you like to be credited?  Is your GitHubDragonFly username okay?

GitHubDragonFly, 2021-02-13:
My username is already in the file and no other credit is required.

I've done my best to turn it into a troubleshooting tool rather than just a simple example.
Whomever is to use it and might need any other features, can try to update it themselves.

GitHubDragonFly, 2021-02-15:
You might consider updating the file I submitted.

Multiple tag entry was using comma but that should conflict with multi-dimensional arrays, ex DINTArray[0,1]

Probably just replace it with semicolon to look like this:

def startUpdateValue():
    global updateRunning

    '''
    Call ourself to update the screen
    '''

    if (path != selectedPath.get()):
        start_connection()
    else:
        displayTag = (selectedTag.get()).replace(' ', '')

        if displayTag != '':
            myTag = []
            if ';' in displayTag:
                tags = displayTag.split(';')
                for tag in tags:
                    if not str(tag) == '':
                        myTag.append(str(tag))
            else:
                myTag.append(displayTag)

            if not updateRunning:
                updateRunning = True
            else:
                try:
                    btnStart['state'] = 'disabled'
                    btnStop['state'] = 'normal'
                    tbPath['state'] = 'disabled'
                    tbTag['state'] = 'disabled'
                    popup_menu_drivers['state'] = 'disabled'

                    results = comm.read(*myTag)
                    allValues = ''
                    if len(myTag) == 1:
                        tagValue['text'] = results.value
                    else:
                        for tag in results:
                            allValues += str(tag.value) + ', '
                        tagValue['text'] = allValues[:-2]

                    root.after(500, startUpdateValue)
                except Exception as e:
                    lbPLCMessage.delete(0, 'end')
                    lbPLCError.insert(1, e)
                    tagValue['text'] = '~'
                    connected = False
                    start_connection()
                    root.after(2000, startUpdateValue)

GitHubDragonFly, 2021-02-15:
There is another change required as well, either set the startup value of myTag to a single list entry or modify the bottom line of this part of code (just as it looks here):

    # create a label and a text box for the Tag entry
    lblTag = Label(frame3, text='Tag(s) to Read', fg='white', bg='navy', font='Helvetica 9 italic')
    lblTag.pack(anchor=CENTER, side=TOP, pady=5)
    selectedTag = StringVar()
    tbTag = Entry(frame3, justify=CENTER, textvariable=selectedTag, font='Helvetica 11', width=80, relief=RAISED)
    selectedTag.set((str(myTag).replace(',', ';'))[1:-1].replace('\'', ''))

GitHubDragonFly, 2021-02-15:
The comment on the top of the file also states "comma" so maybe update that as well and replace it with "semicolon".

GitHubDragonFly, 2021-02-15:
All the above newly mentioned changes were put in a PR on your Develop branch.

GitHubDragonFly, 2021-04-02:
Just in case if anybody might be interested, there is a simple Node-RED project that is using pycomm3 library alongside pylogix:

https://github.com/GitHubDragonFly/Node-RED_PLC_Example