ottowayi/pycomm3 #103: Proper way to handle reconnects?

markovchainz, 2021-02-03:
This is more so an extension off of #60...

Occasionally, a connection to a PLC will get interrupted due to some network issue and the following error is reported back when trying to perform a read `failed to receive reply`.

When the PLC becomes available, the error switches to `failed to send message`.

Handling this error and then performing a `close()` followed by an `open()` does not get communication back up and running.

Is there a better way of handling this situation?

markovchainz, 2021-02-04:
Update:

Re-initializing the handler worked, i.e.:
```
...
except Exception as e:
    # the below obviously needs to be wrapped in a try/except
    plc = LogixDriver('...')
    plc.open()
```

ottowayi, 2021-02-04:
Correct, the drivers can be treated as 'disposable', so if something happens just create a new one.  I would think doing a `close` then an `open` should work, so I'm not sure why that didn't work for you either.  I'll look into it.  
 
One thing that could break is if someone does a download, you will either need to re-create the driver or re-upload the tag list.  Typically, it will use the instance id of tags when making the request to save space in the packet instead of using the full tag name.  Instance ids are generated at download, so if someone does that it will change the ids and could break your driver.  

Some of this stuff could be done automatically, but that would increase the complexity of the driver a lot and could make it not as flexible.  The idea is that the driver is pretty dumb/focused and just handles reads/writes/etc.  More complex features like multithreading, polling, re-connection, etc should be handled by the user to fit their use case.   If I were to add that stuff it would become very opinionated and work how I would like it to... which may not be what everyone else wants.   

Also, I set the connection timeout to a pretty high value.  Something like doing a request every minute or so is enough of a heartbeat to keep the PLC from closing the CIP connection.   But, based on your error message it looks like the socket connection could have closed.  One item on my todo list is to have the tag errors return the exception object, not just a string description.   That would help you in this case to see what that CommError was raised from.

markovchainz, 2021-02-04:
Thanks for the quick response and detailed clarification! 

I'm not too worried about my specific situation not working with open/close alone, it could very well be a network related problem. Troubleshooting certain industrial networking problems are a rabbit hole not worth anyone's time. 

I'll do some more testing but the approach I ended up taking was the following, when an error occurs during a read/write:

1. Attempt to close the connection if `.connected = True`
2. Reinitialize the driver 
3. Attempt to connect

No issues thus far. I also added a minimum of 60 seconds to pass before attempting to reconnect - to prevent constant close/open attempts. I agree with your last part. If there's a way to know that it was a comms error, reconnecting can be done more cleanly. 

I switched from pylogix to pycomm3 because pycomm3 seems more reliable, comprehensive and has SLC support. The only drawback so far has been the reconnects. Pylogix would internally deal with a reconnect attempt when trying to perform a read/write and there was no intervention required by the user. But I agree with your approach and the objective to keep things as simple as possible on the driver side.

closed 2021-02-04 by markovchainz

ottowayi, 2021-02-04:
I think the `.connected` status is sort of misleading, it won't get set false if the connection closes.  Just doing `close()` then `open()` should be fine, there shouldn't be any problem calling close on an already closed connection.   Hopefully soon I'll get to looking into it and seeing if it give a more accurate status.

If you're coming from pylogix, I suggest reading the docs linked in the readme.  There are a number of features that are different, namely the support for complete structure reading/writing and generic messaging.  Part of the work I'm doing right now will make struct and generic message support even more flexible/powerful.   There's nothing in there for SLCs yet, but the original pycomm does have some examples in their readme.

GitHubDragonFly, 2021-02-05:
As for my testing, the app posted in the Pull Requests seems to be handling disconnects ok.

This would be true for as long as there is some sort of communication taking place when the disconnect occurs, like while the tag reading is in progress or either of the discovery of devices or getting tag list was started.

The app is designed for quick testing so it doesn't hurt to keep it handy.