ottowayi/pycomm3 #111: V1.0.0 Release Tests

ottowayi, 2021-02-17:
This issue will be used to track the testing against various devices.   For each device tested against, please leave a comment with the device model and firmware version, the tests performed (i.e.: the built-in ones, user custom scripts/apps, etc), and whether or not they were successful.  Once I feel like the beta releases have been adequately tested and no major issues are found a new stable release will be created.

ottowayi, 2021-02-17:
1769-L23E-QBFC1 - CompactLogix - V20
All included tests
No Issues

coltman151, 2021-02-21:
1756-L71 - ControlLogix - V24
Basic Reads/Writes
No Issues
[pycomm3_V1.0.0_1756-L71_V24_Basic_Read.log](https://github.com/ottowayi/pycomm3/files/6018254/pycomm3_V1.0.0_1756-L71_V24_Basic_Read.log)

coltman151, 2021-02-21:
1756-L72 - ControlLogix - V24
Basic Reads/Writes
No Issues

Attached Log files for these. Ended up just throwing a quick test program together on a different processor. 

[pycomm3_V1.0.0_1756-L72_V24_Basic_Read.log](https://github.com/ottowayi/pycomm3/files/6018296/pycomm3_V1.0.0_1756-L72_V24_Basic_Read.log)
[pycomm3_V1.0.0_1756-L72_V24_Basic_Write.log](https://github.com/ottowayi/pycomm3/files/6018297/pycomm3_V1.0.0_1756-L72_V24_Basic_Write.log)

coltman151, 2021-02-21:
1756-L72 - ControlLogix - V32
Tested with no issues:
- get_plc_info()
- get_plc_name()
- get_plc_time()
- get_tag_info()
- get_tag_list()
- read()
- write()

Not working as tested:
- set_plc_time() (time never changed), log attached.
- open() (docs say it should return true if successful, returns None), log attached
[pycomm3_V1.0.0_1756-L72_V32_Set_PLC_Time.log](https://github.com/ottowayi/pycomm3/files/6018345/pycomm3_V1.0.0_1756-L72_V32_Set_PLC_Time.log)
[pycomm3_V1.0.0_1756-L72_V32_Open.log](https://github.com/ottowayi/pycomm3/files/6018346/pycomm3_V1.0.0_1756-L72_V32_Open.log)

ottowayi, 2021-02-21:
@coltman151 I will look into those couple things that didn't work.  The logs did show that the instance ID part was broken, I think I just pushed a fix for it. See how it has 91 followed by the tag name, that means it's using symbolic addressing.

[image]

coltman151, 2021-02-21:
Open is still returning none for me. Here's the code I'm using just in case you need to see it. 

```
from pycomm3 import LogixDriver
from pycomm3.logger import configure_default_logger, LOG_VERBOSE

configure_default_logger(level=LOG_VERBOSE, filename="pycomm3_V1.0.0_1756-L72_V32_Open_V2.log")

def pycomm_tests():
    with LogixDriver('10.64.138.12/bp/6') as plc:
        return plc.open()

data = pycomm_tests()
print(data)
```

[pycomm3_V1.0.0_1756-L72_V32_Open_V2.log](https://github.com/ottowayi/pycomm3/files/6018394/pycomm3_V1.0.0_1756-L72_V32_Open_V2.log)

ottowayi, 2021-02-21:
oops, was missing a file from that commit

coltman151, 2021-02-22:
I've encountered an issue with the CIPDriver generic_message. Upon sending a message to a Stratix 5700, the only .value I receive back when using the Power Status example is NONE. The current release version returns the binary values like it's supposed to. I've attached the log where you can see the packet coming back with no errors. I'm going to try a few other generic messages I know I can send to a Stratix and see if there's any different results. 

[pycomm3_V1.0.0_Power_Status.log](https://github.com/ottowayi/pycomm3/files/6019419/pycomm3_V1.0.0_Power_Status.log)

coltman151, 2021-02-22:
I ended up grabbing an EDS file for my particular Stratix switch from my work laptop. The link path is being sent correctly `20 F6 24 01 30 01` per the EDS file.

coltman151, 2021-02-22:
Here's what I've tried so far and what I've figured out. 

- setting data_type to `None` returns the bytes b'\xe8\x03\x00\x00'. I'm not sure how to decode this, but the expected value is 1000.
- setting data_type to `'INT'` returns a failed to parse reply error because it can't decode a string
- setting data_type to `INT` returns a NameError because INT is not defined
- setting data_type to `[('test', 'INT'), ]` returns a failed to parse reply error because it can't decode a list
- setting data_type to `[('test', 'INT'), 'INT']` returns same as above

Looking at the definition for data_type, it is looking for `Union[Type[DataType], DataType]`, so I'm not real sure what I need to pass to make it happy. 

In 0.12.0, simply passing `[('test', 'INT'), ] will return the correct value.

Sorry for filling up this thread, if you'd rather keep this to pass/fail comments I can move this to a discussion.

ottowayi, 2021-02-22:
So you just ran into one of the breaking changes.  Part of this new version is a new type system.  These types can be used to represent any CIP data type, array, structure, etc.   The whole library is now based on these types, for example when it uploads the tag list it, it creates a type for that tag which is then used to decode the response on a read or to encode the value when you write.  The current `data_format` arg was renamed to `data_type` and now requires a `DataType` class.  So your 3rd bullet point is the correct one, you just need to import the type.  (`from pycomm3 import INT`).   I haven't written any documentation on this stuff yet,  but it is next on my list and will be before the stable release.

coltman151, 2021-02-22:
Okay cool! I wouldn't have realized I needed to import that class. 

In that case, mark messaging down as good to go as far as I can test at home.

closed 2021-03-10 by ottowayi