ottowayi/pycomm3 #112: Tag List Error

AndrewStuhr, 2021-02-18:
Hey man

I had a script that was successfully reading a PLC's tag list.  I believe someone at my work added addition tags/logic to that PLC's program.  Now I get the following error:

Traceback (most recent call last):
  File "getTagListAB.py", line 6, in <module>
    with LogixDriver(ipPlusSlot, large_packets=False, init_program_tags=True) as plc:
  File "C:\Users\Engineer\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pycomm3\clx.py", line 180, in __init__
    self.get_tag_list(program='*' if init_program_tags else None)
  File "C:\Users\Engineer\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pycomm3\clx.py", line 77, in wrapped
    return func(self, *args, **kwargs)
  File "C:\Users\Engineer\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pycomm3\clx.py", line 603, in get_tag_list
    tags = self._get_tag_list()
  File "C:\Users\Engineer\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pycomm3\clx.py", line 621, in _get_tag_list
    tag['data_type'] = self._get_data_type(tag['template_instance_id'])
  File "C:\Users\Engineer\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pycomm3\clx.py", line 956, in _get_data_type
    return self._cache['id:udt'][instance_id]
KeyError: 2610

Here's the python code:

from pycomm3 import LogixDriver

ip = '192.168.50.48'
slot = '0'
ipPlusSlot = ip + '/' + slot
with LogixDriver(ipPlusSlot, large_packets=False, init_program_tags=True) as plc:
	print(plc.tags)

The PLC is a ControlLogix 1756.  Any ideas?

ottowayi, 2021-02-18:
Interesting.. I haven't seen that happen before.  Did it happen once or keeps happening?  Changes in tags in the controller shouldn't affect the tag list upload since it does a full upload every time.   If you can enable logging, that might help me narrow down the issue.  Also, would you mind trying the new beta version?  It also provides a helper method to enable logging. 

```
from pycomm3.logger import configure_default_logger, LOG_VERBOSE
configure_default_logger(level=LOG_VERBOSE)
```

AndrewStuhr, 2021-02-18:
It happens consistently.  I installed the development branch and ran the following script.  I've attached the output.

from pycomm3.logger import configure_default_logger, LOG_VERBOSE
configure_default_logger(level=LOG_VERBOSE)
[error.txt](https://github.com/ottowayi/pycomm3/files/5999889/error.txt)

from pycomm3 import LogixDriver

ip = '192.168.50.48'
slot = '0'
ipPlusSlot = ip + '/' + slot
with LogixDriver(ipPlusSlot, large_packets=False, init_program_tags=True) as plc:
	print(plc.tags)

ottowayi, 2021-02-18:
What firmware version?  I'm realizing the logging is missing that, so I'll get it added.  I wonder if there is something new or different that I haven't seen before that it's not handling correctly.  I don't see any obvious issue right now, but I'll look at it closer tomorrow and hopefully figure out what the problem is.

AndrewStuhr, 2021-02-18:
Sounds good.  I can ask about the firmware tomorrow.

coltman151, 2021-02-18:
Just out of curiosity, how many tags are in the program, and are there any UDTs? Are there any really big tags, like a DINT[1000] or a huge UDT?

Specifically the extended status `out of memory` is why I ask. 

```
021-02-17 18:40:20,858 [DEBUG] pycomm3.cip_driver.CIPDriver.send(): Received: SendUnitDataResponsePacket(service=b'\x03', command=b'p\x00', error='Destination unknown, class unsupported, instance undefined or structure element undefined (see extended status) - Extended status out of memory  (05, 00)')
Traceback (most recent call last):
  File "C:\venv\lib\site-packages\pycomm3\logix_driver.py", line 715, in _get_data_type
    template = self._get_structure_makeup(instance_id)  # instance id from type
  File "C:\venv\lib\site-packages\pycomm3\logix_driver.py", line 569, in _get_structure_makeup
    raise ResponseError(f"send_unit_data returned not valid data", response.error)
pycomm3.exceptions.ResponseError: ('send_unit_data returned not valid data', 'Destination unknown, class unsupported, instance undefined or structure element undefined (see extended status) - Extended status out of memory  (05, 00)')
```

ottowayi, 2021-02-18:
This is weird, it looks like it's trying to read the data type definition for an instance id of x0a32 - which I don't see as a data type for any tag above.  It makes me think we're not filtering out all of the system tags from the tag list properly.  It also makes me realize I do not have enough trace logs to see exactly what is happening.   Could you try doing this:

[image]
Comment out that exception and return None instead, it's in the `logix_driver.py` file.  And then send me the tag list? 
This is a simple method that I typically use:
```
from pprint import pprint
from pycomm3 import LogixDriver

with LogixDriver('192.168.1.236') as plc:
    with open('c:/tmp/tags.txt', 'w') as f:
        pprint(plc.tags, f)
```

AndrewStuhr, 2021-02-18:
Here's a screenshot of the code change, script, and output:

[image]

ottowayi, 2021-02-18:
dammit,

what about instead of returning `None` doing this:
`self._cache['id:udt'][instance_id] = {'name': 'no-idea'}`

I didn't check to see if there are any keys required later on, but hopefully that will work.   Also, if that works can you print the `plc.info` to a file too?

AndrewStuhr, 2021-02-18:
It didn't work, here's the output:

[image]

I didn't try printing plc.info

ottowayi, 2021-02-18:
ugh... I missed that.
what about: `self._cache['id:udt'][instance_id] = {'name': 'no-idea', 'type_class': None}`

AndrewStuhr, 2021-02-18:
Nice, here's both outputs:

[plc.txt](https://github.com/ottowayi/pycomm3/files/6005077/plc.txt)
[tags.txt](https://github.com/ottowayi/pycomm3/files/6005078/tags.txt)

ottowayi, 2021-02-18:
Perfect, just what I needed.  What is the data type of `AIT515` / `AIT701` / `AIT702` / etc?  That's the type that is failing to upload.

AndrewStuhr, 2021-02-18:
We just went online, with the program and didn't see any of these tags, nor any others with the 'no-idea' property.  However, these tags do resemble a project that another engineer is working on.  I'm wondering if we're getting some cross-readings from another processor or something.  I'm going to try and find where these other tags are coming from...

AndrewStuhr, 2021-02-18:
False alarm.  An engineer removed these tags right before we went online, so it only appeared they were missing.  I should have the tag types soon enough.

GitHubDragonFly, 2021-02-18:
There is a Windows based tag browser that could also be used for quick testing:

https://www.advancedhmi.com/forum/index.php?topic=2835.msg16869#msg16869

It is a standalone app and if it breaks while trying to read tags then there is a problem with tags.

The AdvancedHMI forum requires a FREE registration in order to download this.

AndrewStuhr, 2021-02-19:
Here's the email and pics from my coworker:

Ok I added a tag called AIT515.  It is a tag with data type AOI_SCP_FBD1.  The data type is an add on instruction. I attached the AOI structure and the resultant tag structure.

[image]

[image]

GitHubDragonFly, 2021-02-19:
Should "DBC" be included in the tag?

ottowayi, 2021-02-19:
I would try @GitHubDragonFly's suggestion of that app to make sure it's a pycomm3 bug.  I built the same struct in my PLC just to confirm it works.  Have you tried downloading the program again?  I hate that as a solution, but that might help confirm it's a bug in the library.  I know there was a bug sometime ago with Logix not re-compiling AOIs on download properly unless you did a compact first and cleared the cache.  But I never ran into that issue in the wild, but maybe it could be something similar.

AndrewStuhr, 2021-02-19:
My coworker added the AOI's back into the PLC, and now pycomm3 is connecting normally.  Interesting...

I'm leaning towards some sort of issue on the PLC side.

ottowayi, 2021-03-10:
assuming this is resolved, if not feel free to re-open

closed 2021-03-10 by ottowayi