ottowayi/pycomm3 #123: [BUG] -

tdesroches, 2021-04-12:
**Pre-checks**
- [ x] running latest version
- [ ] enabled logging 
- [x ] consulted the [docs](https://pycomm3.readthedocs.io/en/latest/) 

**Description**
Reading a complex tag returns wrong values when nested structures are too deep

**Target PLC**
Model:  [e.g. 1756-L82ES]
Firmware Revision: [e.g. 31]
Other Devices in CIP Path:  [e.g. 1756-EN2T v11.002]

**Code Sample**
_Minimal reproduceable code sample_
```python
from  pycomm3 import LogixDriver as ld
from pprint import pprint

if __name__ == "__main__":
    with ld('192.168.0.41') as plc:
        #pprint(plc)
        #pprint(plc.info)

        result = plc.read('gstRecipeDownloaded')
        
        for t in result:
            pprint(t)
```

**Additional context**
The UDT I'm reading is complex in that it has several other UDTs and Array nested, so this may simply be outside of the scope of the library.

This is a sample of what I get:
``` 
 'stCamDataX': [{'arrstMotionData': [{'rEndAngle': 36.0,
                                      'rPosition': 1400.0,
                                      'rStartAngle': 324.0},
                                     {'rEndAngle': 216.0,
                                      'rPosition': 1908.0,
                                      'rStartAngle': 144.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0}],
                 'intNumberOfMoves': 2},
 'stCamDataY': [{'arrstMotionData': [{'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 2.444985560553941e-41,
                                      'rPosition': 0.0,
                                      'rStartAngle': -2.0},
                                     {'rEndAngle': 2.436577769767992e-41,
                                      'rPosition': -2.4221443955854463e-41,
                                      'rStartAngle': 2.4253673820533934e-41},
                                     {'rEndAngle': 2.459839324275784e-41,
                                      'rPosition': 2.37548115672343e-41,
                                      'rStartAngle': -2.3939782964525175e-41},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0},
                                     {'rEndAngle': 0.0,
                                      'rPosition': 0.0,
                                      'rStartAngle': 0.0}],
                 'intNumberOfMoves': 0},
```
When I take a read by tunneling a little deeper I get what i expect
```
{'arrstMotionData': [{'rEndAngle': 36.0,
                      'rPosition': 1400.0,
                      'rStartAngle': 324.0},
                     {'rEndAngle': 216.0,
                      'rPosition': 1908.0,
                      'rStartAngle': 144.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0}],
 'intNumberOfMoves': 2}

{'arrstMotionData': [{'rEndAngle': 313.0,
                      'rPosition': 675.0,
                      'rStartAngle': 266.0},
                     {'rEndAngle': 94.0,
                      'rPosition': 472.0,
                      'rStartAngle': 46.0},
                     {'rEndAngle': 0.0,
                      'rPosition': 1172.0,
                      'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0},
                     {'rEndAngle': 0.0, 'rPosition': 0.0, 'rStartAngle': 0.0}],
 'intNumberOfMoves': 2}
```

As you can see it seems to read the first instance of the structure but seems to fall apart afterwards.
I've attached a screen shot of part of the structure.

[image]

ottowayi, 2021-04-12:
yeah this is a definitely a bug and I have a feeling I know what it is.  In Logix all the structs are padded to multiples of 4-bytes and I think I didn't account for that in the parsing and my complex UDT test doesn't require any padding.  I see you an have `intNumberOfMoves`, if that is an `INT` and if there aren't any other attributes that use 2 bytes, your UDT is padded with 2 extra bytes. So it parses the first UDT correctly, but leaves the padding in the stream which is then parsed as part of the next UDT.  I'm going to test it out and get back to you with what I find.

tdesroches, 2021-04-12:
Ok thanks!  intNumberOfMoves is techanically a SINT, not sure if that matters.  If you need anything from me let me know.

ottowayi, 2021-04-12:
Yep I believe that was it.  I was able to replicate the issue and it was a simple change to fix it.  I pushed it to the develop branch, you can either pull from there or modify `pycomm3/custom_types.py` to add this line:
[image]

Thanks for finding this, I updated the tests to include this scenario.  I'm currently working on a new release that will hopefully be out really soon.

tdesroches, 2021-04-12:
I found 2 more similar maybe related issues?  I added the line to my source files as below.  This seems to have fixed the orginal issue.
[image]

I can make these two seperate issue if you'd prefer.

1) Similar issue reading Same stRecipeDownloaded structure but different part
GOT:
[image]
Expected:
[image]

2) Extra Bytes around string?
There is a string at the top level of the stRecipeDownload.  I can see the name J1305 but its padded front and rear. 
[image]

I'm not sure if these are related,  I can make a seperate Issue for each if you'd prefer.

ottowayi, 2021-04-12:
hmm.. the first one is definitely related and the second may or may not be, so keeping this issue is fine.  If possible, could you send me the UDT so I can try testing against it?  My email is on my profile if you don't want to attach it here.

tdesroches, 2021-04-12:
Yup no problem, I'll send it now.  On the bright side I don't think you'll encounter a more complex UDT lol

ottowayi, 2021-04-12:
I think I got it, sometimes there is padding added to the host SINTs for BOOLs inside the UDTs.  I changed the parsing to make sure it starts decoding at the correct offset for each attribute, discarding any padding between them.  It wasn't a hard fix, but difficult to locate.  I pushed the change to the `develop` branch.

tdesroches, 2021-04-12:
Awesome, tested it and all looks good!

ottowayi, 2021-04-12:
Great! thanks again for reporting, perfect timing too.  I'm hoping to put out the next release this week, so keep an eye out for it.

tdesroches, 2021-04-12:
I haven't tried it yet but could this issue also apply for writing it back?

--
Tony Desroches
Programmer - Controls
Linear Transfer Automation

P. 705-735-0000 ext. 251
C. 705-627-9345
F. 705-735-4688
E. ***@***.***> ***@***.******@***.***>

On Mon, 2021-04-12 at 12:34 -0700, ottowayi wrote:

Great! thanks again for reporting, perfect timing too. I'm hoping to put out the next release this week, so keep an eye out for it.

ottowayi, 2021-04-12:
It absolutely does and I totally forgot to fix that as well. I'll try and fix that tonight too.

tdesroches, 2021-04-12:
I should be in the office and can test it when its ready.

Tony Desroches
Controls Programmer
Linear Transfer Automation Inc.
C: 705-627-9345
F: 705-735-4688
E: ***@***.***

Confidentiality Notice:
This transmission (including any attachments) may contain confidential information belonging to the sender and is intended only for the use of the party or entity to which it is addressed.  If you are not the intended recipient, you are hereby notified that any disclosure, copying, distribution, retention or the taking of action in reliance on the contents of this transmission is strictly prohibited.  If you have received this transmission in error, please immediately notify the sender and erase all information and attachments.

ottowayi, 2021-04-12:
I re-opened this issue until I for sure have a fix done.  I just pushed a change that should fix it, but I only did a quick test.  I still need to add unit tests and make sure it's actually fixed.  I might get back to it later tonight, if not I should be able to finish it tomorrow.

ottowayi, 2021-04-13:
I believe this issue is resolved, please re-open if you discover otherwise.  I've merged all changes to master and will be putting a new release out within the next couple days.

closed 2021-04-13 by ottowayi