ottowayi/pycomm3 #127: [BUG] -

AndrewStuhr, 2021-04-23:
I've got 2 interesting issues reading/scanning from a SLC 505:

1. There's an O1 array that isn't getting scanned.  However, O0 is getting scanned (tagOutput.txt)

2. We can read some of the tags from O1, specifically O1/0 through O1/30.  But we can't ready O1/31 and higher. (readOutput.txt)

[info.txt](https://github.com/ottowayi/pycomm3/files/6367721/info.txt)
[readOutput.txt](https://github.com/ottowayi/pycomm3/files/6367722/readOutput.txt)
[readSLCsimple.txt](https://github.com/ottowayi/pycomm3/files/6367723/readSLCsimple.txt)
[tagOutput.txt](https://github.com/ottowayi/pycomm3/files/6367724/tagOutput.txt)

Let me know if you have any thoughts!

GitHubDragonFly, 2021-04-26:
You can try using the following Windows based app, which is using libplctag library, for testing what it can read from your O:1:

https://github.com/GitHubDragonFly/AB-Modbus-Master

Your O:1 addressing looks like as if targeting expansion modules.

ottowayi, 2021-04-26:
Hmm sorry I'm not too familiar with SLC addressing, my work always dealt with replacing them and rarely did anything beyond simple logic bugfixes.  Could you also share some screenshots using the RSLinx Data Monitor so I can see what it is supposed to look like?  The full file list and the O files you're trying to read would help, bonus points if you can get me a wireshark capture during it as well.  I could then compare the requests I'm making with the ones RSLinx is and look for differences.

acedrew, 2021-05-05:
Sorry I neglected to read this issue because the title didn't click, but I believe #138 may solve this as well. The addressing you're showing is implicit position, which the SLCDriver did not handle correctly.

acedrew, 2021-05-06:
@AndrewStuhr can you check out #138?

closed 2021-07-03 by ottowayi