ottowayi/pycomm3 #134: Can't make the read from PF525 example work.

armandom84, 2021-04-29:
I think I'm having issues with the path, as I always get this response 

"pf525_param, None, INT, Service not supported"

and I think my issue is the path, so any help will be appreciated. 

My path is : PLC/backplane/2/enet/VFD 
'172.16.169.162/backplane/2/enet/172.16.169.13'

I have tried changing the 2 without much success. 

Regards

ottowayi, 2021-04-29:
Hmm, could you post a screenshot of the path to the drive in RSLinx?

ottowayi, 2021-04-29:
I noticed you also posted a question in pylogix and said your PLC in a CompactLogix.  I think the newer versions use different port numbers, `enet` is a simple replacement for `1`.  Can you try replacing it with a `3` or `4` depending on which port you're using?  https://www.plctalk.net/qanda/archive/index.php/t-109146.html

closed 2021-07-03 by ottowayi