ottowayi/pycomm3 #139: SLCDriver ReWrite Requirements

acedrew, 2021-05-06:
@ottowayi can this serve as a thread with the other SLC users and yourself to define requirements/needed improvements for the SLCDriver? I may be able to point some resources toward working on this in the near future, or tackle it myself.

- Test performance gains of eliminating fall through full REGEX matching, instead matching just the file name and directing to the correct full length match.
- Add more descriptive error messages and better handling to the tag parser
- Automatic pipelining of adjacent read requests
- Other structural changes you want to see?

ottowayi, 2021-05-06:
Yeah this looks good.

Off the top of my head right now I can think of:

- change tag parsing to a single regex and then validate based on type
- add SLC packet classes to move more code out of the individual driver methods
- identify features exclusive to SLC and MicroLogix (each family) and determine if the driver should be separated into 2 different ones
- add unit tests and documentation

GitHubDragonFly, 2021-05-07:
This is sort of an FYI and related to MicroLogix 1100.

The attached py files were modified to include changes from PR https://github.com/ottowayi/pycomm3/pull/138 as well as new additions for reading PID values from ML1100.

Example addressing:
- PD11:0 will show a list of 23 INT values
- PD11:0.LUT will show INT value for Loop Update Time
- PD11:0.EN will show bit/bool value

The text file only shows ML1100 PID structure.

[ML1100 PID.txt](https://github.com/ottowayi/pycomm3/files/6439436/ML1100.PID.txt)

[py files.zip](https://github.com/ottowayi/pycomm3/files/6439431/py.files.zip)

acedrew, 2021-05-10:
@GitHubDragonFly would you be interested in working on this and a few other issues we have? We'd love to sponsor getting this knocked out, especially figuring out some other weirdness around I/O file indexing.

ottowayi, 2021-05-10:
I just created a `slc_redo` branch we can use for this, I think you guys should be able to push to it?  

I have some ideas of where I'd like to start and what tasks should be done first, but I haven't started any of the work.  I think best place to start would be to make a common SLCPacket class, subclassed from SendUnitData (I think), that would have the common code for every request.  Then go into making read/write packets.  At the same time I'd totally start over with the tag request parsing since it's a nightmare to follow.

GitHubDragonFly, 2021-05-11:
I will pass on collaborating on this since, in my opinion, the SLC driver seems to work properly in general.

If I do come up with any new suggestions then I will let you know and could also test whatever changes you might make.

acedrew, 2021-05-11:
@GitHubDragonFly We're struggling to figure out the best way to parse I/O Tags that are indexed logically in the pycomm driver (word location in I file) but are indexed by physical card layout in most SCADA systems. Any ideas there?

closed 2023-03-26 by ottowayi