ottowayi/pycomm3 #141: plc.tags JSON formatting

AndrewStuhr, 2021-05-13:
Hello!  There appears to be an issue with the output of plc.tags in pycomm3 1.1.1.  It's not formatting to json like it was in previous version.  The following script produces the issue:

from pycomm3 import LogixDriver
import json

with LogixDriver('192.168.100.54', large_packets=False, init_program_tags=True) as plc:
    print(json.dumps(plc.tags))

Here's the error message:

[image]

If you could take a look, it would be much appreciated.  Thanks for all your work!

ottowayi, 2021-05-13:
That's because the new type system was introduced and those objects are not JSON serializable.  Would adding a separate method be acceptable for a JSON-compatible data structure?  It shouldn't be hard to do and I could include it in the next version.

In the mean time, if you filter out the `type_class` keys it should work how it used to.

AndrewStuhr, 2021-05-13:
Thanks!  Yep I see the type_class properties now.  If it's not too much work, adding the JSON method would be helpful.

GitHubDragonFly, 2021-05-14:
Since you have mentioned JSON then just to throw this in here in case if anybody might find it useful.

There is this [pylogix-api](https://github.com/TheFern2/pylogix-api) project which is using Flask and JSON.
One could use some ideas from it or modify it to use pycomm3 instead or alongside pylogix.

closed 2021-07-03 by ottowayi