ottowayi/pycomm3 #143: [BUG] - Can't read the tag values using alias name

ismaan1998, 2021-05-21:
**Description**
I am using Micro800 emulate to test the script.  
[image]

The output 00 (tag name = _IO_EM_DO_00 and alias name =  db_status)

code = 
```` 
with LogixDriver("192.168.43.150") as plc:
        print(plc.read("_IO_EM_DO_00")) 
````
If I read the value using the tag name , Its working fine. 

But I want to read the value using tag alias. Is it possible ?  
example - 
````
with LogixDriver("192.168.43.150") as plc:
        print(plc.read("db_status"))
````
Output - 
````
db_status, None, None, Tag doesn't exist - db_status
````

ottowayi, 2021-05-21:
Hmm.. so this looks like another special case for these stupid controllers.   What is happening is when pycomm3 first connects, it uploads all the tags in the PLC stores them in the driver.  Then when you go to read the tag it looks up the info in the tag list so it can properly make the request, but if the tag is not in that list you will get this error.   It looks like the Micro800 aliases are not returned in the tag list like how all the other Rockwell PLCs do.  I also discovered these also will exclude any array tags you have that don't start at index 0.    

I haven't used the Micro800s before besides the simulator for testing, can you use aliases in an HMI?  If so I may be able to wireshark Linx and figure out a workaround.  

I did find a temporary solution, you can create your own 'alias' in pycomm3 by doing:
```
with LogixDriver("192.168.43.150") as plc:
    plc.tags['db_status'] = plc.tags['_IO_EM_DO_00']
    print(plc.read('_IO_EM_DO_00', 'db_status')) 
```

```
[Tag(tag='_IO_EM_DO_00', value=False, type='BOOL', error=None), Tag(tag='alias_tag', value=False, type='BOOL', error=None)]
```

You will also have to modify the `tag_request_path` method in `pycomm3/packets/util.py` with this change:
[image]

All of my tests passed, so I think I will keep that change and include it in the next release.  For now though this should at least let you manually add the aliases.

ismaan1998, 2021-05-25:
Thanks, that worked