ottowayi/pycomm3 #145: pycomm3.exceptions.ResponseError: get_plc_info did not return valid data - Service not supported

lemorn, 2021-05-28:
Hello！When I ran the test code:

from pycomm3 import LogixDriver
with LogixDriver('192.168.1.118/1') as plc:
    print(plc)
    print(plc.info)

A error occurred：
[image]
 And I captured all the traffics:
[image]

My PLCs are:
[image]

Thanks very much!

GitHubDragonFly, 2021-05-28:
You should be using SLCDriver instead, which usually requires IP only (without path).

lemorn, 2021-05-30:
Thank you! I get the infomation from my PLC. But I find some new questions: what's the means of the revision,status and state? 
If I want to know what mode the PLC is in(Reomte,Run or Program)  and whether a password is used ，what should i do?
How can I construct my own data packet and send it to 1100 or 1400 for operations such as changing their mode?
Thanks very much！
[image]

GitHubDragonFly, 2021-05-30:
Those questions are for @ottowayi while I don't think that the mode related functionality is available in the SLCDriver.

ottowayi, 2021-05-31:
The `list_identity` method returns the standard CIP Identity Object parsed into a dict. The revision is the firmware version of the PLC, so yours is running 2.21. The status is a 2-byte value with different bits meaning different things.  In the LogixDriver it's parsed to give you the keyswitch position, but I haven't added that for the SLCDriver yet.  I'm guessing that it is going to be slightly different for each family, SLC, 1100, 1400, etc, or it could be exactly the same.  I haven't investigated that part yet for the SLCDriver devices.  I'm forgetting what the state is exactly, I want to say it is running/faulted/etc.  The SLCDriver is going to eventually be re-written so that I can support it better, right now it's kind of a mess and I did the bare minimum to get it working from the original pycomm library. 

As for the modes, I don't have anything built-in for that (yet?).  If you can get/set the mode thru a MSG instruction, you should be able to use the `generic_message` method to achieve the same.  [Here](https://docs.pycomm3.dev/en/latest/usage/cipdriver.html#generic-messaging) is a link to the docs that explain how to use generic messaging.  

I'm in the process of moving right now, so my PC with all my pycomm3 stuff as well as my PLCs are still at my old place.  That's why I didn't try it out before responding.  It will be a couple weeks before I have some down time to dig into it, but if you get something working before then, I'd be willing to include it in the docs/examples or possibly include it in the driver itself.

GitHubDragonFly, 2021-05-31:
@ottowayi when you do find time then see also if this could help you any:

https://www.advancedhmi.com/forum/index.php?topic=2822.0