ottowayi/pycomm3 #147: Will this work with Festo EtherNet/IP interface?

jamesbraza, 2021-06-04:
Hello pycomm3!

I am a software engineer in a biotech startup, and we are using a Festo EtherNet/IP interface ([CPX-AP-I-EP-M12](https://www.festo.com/us/en/a/8086610/?q=CPX-AP-I-EP-M12)) in a custom instrument.  I am not a new developer, but I am totally new to EtherNet/IP.

The addresses of every parameter can be found in Section 9 Parameterisation of [CPX-AP-I-EP-M12-EN Operating - EtherNet/IP interface](https://www.festo.com/net/en_us/SupportPortal/Downloads/629777/696286/CPX-AP-I-EP-M12_instruction_2019-08_8099750g1.pdf).

I see in https://github.com/ottowayi/pycomm3/issues/17 that this library is specific to Rockwell PLCs.

Can I confirm that I will not be able to use this library for generic EtherNet/IP comms with a Festo device?
- If yes: can you point me to a relevant example?
- If no: can you provide any pointers on where to get started?

ottowayi, 2021-06-04:
You should be able to use generic messaging to get it to work with this device.   You will want to use the `CIPDriver`, the other 2 drivers are specific to Rockwell's PLCs. [This section](https://docs.pycomm3.dev/en/latest/usage/cipdriver.html#generic-messaging) in the docs talks about how to use the `generic_message` feature.  The Festo docs seem to have everything you need in them to build the message.  You may also want to read over the [data types](https://docs.pycomm3.dev/en/latest/usage/cipdriver.html#generic-messaging) section too, that might help if you want to access some of the more complicated structure objects.

ottowayi, 2021-06-04:
Here is an example I threw together to get attribute 3 ( MAC address) from the Ethernet Link object.  (Disclaimer, I typed this in the comment and did not try it out first). 

```
from pycomm3 import CIPDriver, Struct, UINT, Services, 

with CIPDriver('ip.addr.of.device') as driver:
    result = driver.generic_message(
         service=Services.get_attribute_single,
         class_code=0xF6,
         instance=1,   # TP1 = 1, TP2 = 2
         attribute=3,
         data_type=UNIT[6],
         name='read_mac'
    )
   
    print(repr(result))
```

GitHubDragonFly, 2021-06-04:
I have tried your code with MicroLogix 1100 and modified it to actually get all attributes for 2 codes:

`from pycomm3 import CIPDriver, Struct, UINT, Services

with CIPDriver('192.168.1.10') as driver:
    result_0xF5 = driver.generic_message(
         service=Services.get_attributes_all,
         class_code=0xF5,
         instance=1
    )

    print(repr(result_0xF5))

    if not result_0xF5.value is None:
        print(str(len(result_0xF5.value)))
        print(':'.join(str(x) for x in result_0xF5.value))

    result_0xF6 = driver.generic_message(
         service=Services.get_attributes_all,
         class_code=0xF6,
         instance=1
    )

    print(repr(result_0xF6))

    if not result_0xF6.value is None:
        print(str(len(result_0xF6.value)))
        print(':'.join(str(x) for x in result_0xF6.value))`

GitHubDragonFly, 2021-06-04:
I am still experimenting with this but the `get_attribute_single` in your code seems to have issues parsing the reply while it does show 6 values if it is replaced with `get_attributes_all`.

GitHubDragonFly, 2021-06-04:
All my comments are intended for @ottowayi for any future MicroLogix related updates and if they can help @jamesbraza then even better.

This is the code that worked for reading IP Address by using USINT instead of UINT and with `get_attribute_single`:

from pycomm3 import CIPDriver, Struct, USINT, UINT, Services

with CIPDriver('192.168.1.10') as driver:
    result_0xF5 = driver.generic_message(
         service=Services.get_attribute_single,
         class_code=0xF5,
         instance=1,
         attribute=5,
         data_type=USINT[4],
         name='read_IP'
    )

jamesbraza, 2021-06-06:
Alright!  Thanks to the given sample code, I got very excited to use pycomm3.

I played around a _bunch_ today, and figured out using `Services.get_attributes_all` at `ClassCode.identity_object` I can get the response!  Here's the raw values: `b'\x1a\x00\x0c\x00\xe96\x01\x024\x00\xa7<\x00\x00\x0bCPX-AP-I-EP\x03\x00\x00\x00'`.

I then created a custom `Struct` to decode the Identity Object.

<details>
<summary>My code, if you're curious</summary>
<p>

```python
"""Playing around with pycomm3."""

from __future__ import annotations

from dataclasses import dataclass
from io import BytesIO
from typing import Any, ClassVar, Dict

from pycomm3 import (
    PRODUCT_TYPES,
    SHORT_STRING,
    STATES,
    UDINT,
    UINT,
    USINT,
    VENDORS,
    CIPDriver,
    ClassCode,
)
from pycomm3 import Revision as RevisionDataType
from pycomm3 import Services, Struct, Tag, n_bytes

CPX_AP_I_IP_ADDRESS = "1.2.3.4"

class IdentityObject:
    """Identity (ID) Object."""

    @dataclass
    class Revision:
        """Revision field of the Identity Object."""

        major: int = -1
        minor: int = -1

    _DataType = Struct(
        # Instance attributes
        UINT("vendor_id"),
        UINT("device_type"),
        UINT("product_code"),
        RevisionDataType("revision"),
        n_bytes(2, "status"),
        UDINT("serial_number"),
        SHORT_STRING("product_name"),
        USINT("state"),
    )

    class DataType(_DataType):
        """Data type for the Identity Object."""

        @classmethod
        def _decode(cls, stream: BytesIO) -> Dict[str, Any]:
            """Decode the values in the response to a dictionary of objects."""
            values: Dict[str, Any] = super()._decode(stream)
            values["vendor_id"] = VENDORS.get(values["vendor_id"], "UNKNOWN")
            values["device_type"] = PRODUCT_TYPES.get(values["device_type"], "UNKNOWN")
            values["state"] = STATES.get(values["state"], "UNKNOWN")
            values["revision"] = IdentityObject.Revision(
                values["revision"]["major"], values["revision"]["minor"]
            )
            return values

    CLASS_CODE: int = ClassCode.identity_object
    SERVICE: bytes = Services.get_attributes_all

    # Class attributes
    revision: ClassVar[int]
    max_instances: ClassVar[int]
    num_instances: ClassVar[int]
    max_num_class_attrs: ClassVar[int]
    max_num_instance_attrs: ClassVar[int]

    def __init__(self, **kwargs):
        self.vendor_id: int = kwargs["vendor_id"]
        self.device_type: int = kwargs["device_type"]
        self.product_code: int = kwargs["product_code"]
        self.revision: IdentityObject.Revision = kwargs["revision"]
        self.status: bytes = kwargs["status"]
        self.serial_number: int = kwargs["serial_number"]
        self.product_name: str = kwargs["product_name"]
        self.state: str = kwargs["state"]

    def __repr__(self) -> str:
        return str(self.__dict__)

with CIPDriver(CPX_AP_I_IP_ADDRESS) as cip_driver:
    result: Tag = cip_driver.generic_message(
        service=IdentityObject.SERVICE,
        class_code=IdentityObject.CLASS_CODE,
        instance=1,
        data_type=IdentityObject.DataType,
        connected=False,
    )
    identity_object = IdentityObject(**result.value)
```

</p>
</details>  

I noticed the response only contains the bytes for the instance attributes of the Identity Object.

How can one get the class attributes of the Identity Object?  @ottowayi @GitHubDragonFly

ottowayi, 2021-06-06:
I'm on my phone right now, so I can't confirm, but I think instance 0 will get you the class itself.

GitHubDragonFly, 2021-06-06:
@jamesbraza your code works for me only if I remove the "state", which by the way is not listed as a part of Identity Object on this page:

https://docs.pycomm3.dev/en/latest/cip_reference.html#cip-services-and-class-codes

This is just a note for anybody else who might try your code.

jamesbraza, 2021-06-06:
Thank you @ottowayi, that did the trick!  So far, this library indeed works with my Festo device, so I will close this ticket.

Thank you again!

closed 2021-06-06 by jamesbraza