ottowayi/pycomm3 #149: Use of IdentityObjectInstanceAttributes from object_library

jamesbraza, 2021-06-06:
I see in the `object_library.py` file there are some useful classes like `IdentityObjectInstanceAttributes`.  I read around in the example files, the docs, and the library's source code, and don't see how it gets used.

Does `IdentityObjectInstanceAttributes` key into the library somehow?  Can I use this to query an entire Identity Object?

Or is it around just as a developer convenience to be `import`ed into an end user's source code?

ottowayi, 2021-06-07:
Ummm good question.  I think I was going to use that in some way and ended up not.  I guess it kind of works as importable documentation? You could use it to get the attribute number and data type if you're building a message for the IndentityObject.  I don't see much use for it, so I'm going to delete it and move that information to the docs.

jamesbraza, 2021-06-08:
Okay thank you for confirming @ottowayi.  This was just a question ticket, so feel free to close it, unless you want to keep it open as a reminder.

closed 2021-07-03 by ottowayi