ottowayi/pycomm3 #152: [BUG] - CIPDriver.generic_message - return incorrectly type hinted

jamesbraza, 2021-06-11:
**Description**

The return of `CIPDriver.generic_message` is a `Union[Tag, ResponsePacket]`.

This is an incorrect type hint.  For why, please see the below code sample:

```python
result: Tag = cip_driver.generic_message(...)
```

`mypy` would correctly complain about this, saying:

```
error: Incompatible types in assignment (expression has type "Union[Tag, ResponsePacket]", variable has type "Tag")  [assignment]
```

**Posing Potential Bugfix**

It's my understanding that the `generic_message` method returns:

- `Tag`: if the arg `return_response_packet=False`
- `ResponsePacket`: if the arg `return_response_packet=True`

The correct way to type hint this would be to use `typing.overload` ([docs](https://docs.python.org/3/library/typing.html#typing.overload)) paired with `typing.Literal`.  Please see [this stack overflow](https://stackoverflow.com/a/59361700/11163122) answer to add color as to what an implementation would look like.

One dilemma: pycomm3 support Python 3.6+, and `typing.Literal` is Python 3.8+.  There is a `typing-extensions` library (authored by one of [PEP 483](https://www.python.org/dev/peps/pep-0483/)'s authors, who is also a core mypy contributor) that backports `typing.Literal`.  However, I also observe pycomm3 has zero external dependencies.  I am not sure if pycomm3 is open to adding third party dependencies.

**Known Workaround**

One can use `typing.cast` as a workaround, in the meantime.  This is what I have done, so no pressure to actually fix this type hint.

```python
from typing import cast

result = cast(Tag, cip_driver.generic_message(...))
```

**Additional context**

Versions

```
Python==3.8.8
mypy==0.812
pycomm3==1.1.1
```

ottowayi, 2021-06-11:
Thanks, I didn't know that would cause an issue with mypy.  Since the response packet thing really is just a convenience/hack for internal use, maybe I should make it just return a Tag object.  For the cases when I need the response packet I can use that flag to set the Tag.value to the packet. I could even hide that flag in the **kwargs so it doesn't cause any confusion.

jamesbraza, 2021-06-11:
Yeah, both of you proposed solutions sound good to me! :)

ottowayi, 2021-07-03:
changed pushed to the develop branch, will be included in the next version

closed 2021-07-03 by ottowayi