ottowayi/pycomm3 #153: Writing to the force mask tags within Studio 5000

gary-hsu, 2021-06-21:
Hey, I was wondering if there was a way to use the API to access the force masks in Studio 5000. I got the library connected to an Allen Bradley plc and successfully wrote values, but I want to be able to force values to certain parts using Python and can't figure it  reading the documentation so far.

Thanks in advance!

ottowayi, 2021-06-22:
Interesting, I'm not sure if that is possible.  I just checked in wireshark and I think I found the requests, but I haven't actually trying to decode the message. I will take a closer look at it later to see if it's something I can add. It's not part of the CIP spec so it would require reverse engineering, I've had success with doing it for some features but failed for others.

gary-hsu, 2021-06-22:
I seriously appreciate it. I've been lookin for this feature in multiple different libraries (even non-python ones) and haven't found much success in finding this feature. I know it's possible because I've used software that successfully performed force masks using code, I just can't for the life of me find out how. Let me know your findings, you have my eternal gratitude!

tlf30, 2023-07-04:
@gary-hsu forces require a privileged connection. If implemented, forces would only work on v19 and below of control logix.