ottowayi/pycomm3 #158: [BUG] - clx Writing multiple times different values to a same tag only writes first value

cpcga, 2021-07-02:
**Description**
If we write a tag with the same connection session several times with different values in the target PLC we only see the first value of the batch.

It seems that connection is using some kind of cache

**Target PLC**
Model:  1769-L30ER
Firmware Revision: 32.013

**Code Sample**

#This code fails
import time        
 with LogixDriver('10.146.35.86/0') as plc:
            test_dint = plc.read('test_dint').value
            test_dint = test_dint+1
            plc.write('test_dint', test_dint)
            time.sleep(5)
            test_dint = test_dint + 1
            plc.write('test_dint', test_dint)
            time.sleep(5)
            test_dint = test_dint + 1
            plc.write('test_dint', test_dint)

#This code is working properly (openig and closing connection between each write)
        with LogixDriver('10.146.35.86/0') as plc:
            test_dint = plc.read('test_dint').value
            test_dint = test_dint+1
            plc.write('test_dint', test_dint)
            plc.close
            time.sleep(5)
            test_dint = test_dint + 1
            plc.open()
            plc.write('test_dint', test_dint)
            plc.close()
            time.sleep(5)
            test_dint = test_dint + 1
            plc.open()
            plc.write('test_dint', test_dint)

**Additional context**
 - test_dint is a DINT clx type

ottowayi, 2021-07-02:
Umm, yeah that is definitely a bug.  I'm not sure where though, because no state should be stored between calls.  I will look into it.

ottowayi, 2021-07-02:
[image]
This might take some more research, because it shows that it's writing the second value... but I'm having the same behavior on my PLC too.

ottowayi, 2021-07-03:
I located the issue, as part of a previous bugfix the sequence number stopped getting updated between requests.  I think that caused the PLC to ignore any requests after the first because the sequence number wasn't increasing.  I pushed the change to the develop branch.  Unfortunately I have plans tonight and won't be able to put out a new release. I have a few more things I'd like to do before the next release, but hopefully I'll get them done soon.  In the mean time you can install from the develop branch and watch for a new release.

closed 2021-07-03 by ottowayi

cpcga, 2021-07-03:
Thank you very much for your quick answer.