ottowayi/pycomm3 #164: [BUG] - erro (A session must be registered before a Forward Open)

Gullit1234, 2021-08-30:
[image]

hello, I'm trying to exercise a logic that when inserting a data entry (turns on) or (turns off) the python sends this command to the clp, but when it is giving an error.

if the command is direct, it executes correctly, no problem.

plc.white(B1,True) ok
bad if I make an entry of data and conditions and then method gives an error

ottowayi, 2021-08-30:
What is happening is you are defining the `escreve` method under the `with LogixDriver` block,  but later on you're calling it outside of that block.  Using the `with` statement, it opens a connection at the beginning, runs the code underneath it, and the closes the connection automatically.  So when you go to call `escreve` later on the connection has already been closed, that's why you see that error.  What you can do instead is:

```
ip = str(input(.....
plc = LogixDriver(ip, init_info=False)

def escreve():
    plc.write('B1', ?)   # also you need a value here to write as well

# you should not be calling `_forward_open` yourself, the `open` method does that for you. 
plc.open()  # open the connection yourself without using `with`

... # the rest of your code

plc.close()  # close the connection manually
```

ottowayi, 2021-10-11:
no response

closed 2021-10-11 by ottowayi