ottowayi/pycomm3 #165: Logging In Pycomm3 Stopping Try Except?

JL00001, 2021-09-01:
I have this bit of code that scans a rack-based PLC and prints off what it finds. If there is an empty module the try-except will pass over the error and append a None value to a list and it works great. See Capture1 

[image]

Sample code for Capture1
```python
def get_module_info(slot):
    driver = pycomm3.CIPDriver("192.168.8.38")
    driver.open()
    try:
        data = driver.get_module_info(slot)
        driver.close()
        return(data)
    except pycomm3.exceptions.ResponseError:
        return None

list = []
for x in range(4):
    list.append(get_module_info(x))

print(list)
```

However, when I use it in my code-base; with a logging setup already; the try-except does not catch the error and pass over it. It generates an error message through logging and doesn't run the except code. See Capture2. I can include my code-base but it is sizeable and would probably raise more questions than it answered.

the last 3 lines are messages created by pycomm3 logging integration and it seems to be runing that rather than the try-except.

[image]

My question is why is having an existing logging setup preventing my try-except from running and how can i fix it?

ottowayi, 2021-09-02:
Hmm.. I don't see how logging could be interfering here.  I'm unable to recreate the issue with the code sample you provided.  Have you stepped thru with a debugger and verified the exception is being caught elsewhere?  If you can share more code that would help, if you don't want to post it here my email is on my profile.  

Also, I would move the `driver.close()` call into a `finally:` block after the `except`.  That way the driver is closed every time, not just when there isn't an error.

JL00001, 2021-09-02:
Ok, I was a idiot. the code was running; I just wasn't smart enough to understand my own code. It still generates a error message but the try-except-finally is working. Is there a way to silence error reporting for a few lines to keep it from generating the error messages and then re-enable it?

ottowayi, 2021-09-02:
lol no worries, I'd rather have you open a non-issue than not open an issue for an actual problem.  

Would something like this work?
```python
import logging

def get_module_info(slot):
    driver = pycomm3.CIPDriver("192.168.8.38")
    driver.open()
    logging.disable()

    try:
        data = driver.get_module_info(slot)
    except pycomm3.exceptions.ResponseError:
        data = None
    finally:
        driver.close()
        logging.disable(logging.NOTSET)
        return data

list = []
for x in range(4):
    list.append(get_module_info(x))

print(list)

```

ottowayi, 2021-09-02:
Also, you can use a `with` block so that the `open`/`close` is handled automatically like:

```python
def get_module_info(slot):
    with pycomm3.CIPDriver("10.1.128.45") as driver:
        try:
            logging.disable()
            data = driver.get_module_info(slot)
        except pycomm3.exceptions.ResponseError:
            data = None
        finally:
            logging.disable(logging.NOTSET)
            return data

list_ = [get_module_info(x) for x in range(4)]

print(list_)
```

JL00001, 2021-09-02:
Ok. I saw that function on the logging documentation. Could not get it too work as described. With your mention of it I did some more investigation of the function and discovered 2 things:

Thing 1:
I was having trouble invoking the method so I did a print(dir()) of my logging object

[image]

So it is there but when I try to invoke it I get this error

[image]

Apparently disable is a BOOL? I assume you read the same manual I did and saw the same code and that was not how it was documented. Anyway...

Thing 2:
I ran with disable being a BOOL and tried to set it TRUE and FALSE as a way to disable the logging and it kind of worked

[image]

This code above produced this output below

[image]

As you can see the logging of "Line 420" and "Line 430" is suppressed because the logging is disabled. But the pycomm3 messages still are produced. 

I think I need to redo my logging setup as there appears to be more than one logging object running in my code. I am copying the logging object to my class and referencing that everywhere and I just need to not do that

JL00001, 2021-09-03:
```python
    def __get_module_info(self, slot):
        with pycomm3.CIPDriver(self.ip) as driver:
            logging.getLogger("pycomm3.cip_driver.CIPDriver").disabled = True
            try:
                data = driver.get_module_info(slot)
            except pycomm3.exceptions.ResponseError:
                data = None
            finally:
                logging.getLogger("pycomm3.cip_driver.CIPDriver").disabled = False
                return data

```
This bit of code worked for me to suppress the pycomm3.cip_driver.CIPDriver logging message. As to why the .disable() is a BOOL and not a method as the documentation I will leave you and the 10 or so people that will ever read this to ponder. 

Lessons Learned: The pycomm3 logging is a different logger than I was using, so to modify it you need to specify it using the logging method  logging.getLogger() and set the value to TRUE or FALSE to disable it.

 
To ottowayi, Thanks for your help and for creating this great library. It really helped last summer when I was on site deploying about 500 PoweFlex-525. They were setting the parameters by hand and it took five of them a week to deploy them all. Once I got a script working, it took 5 mins. Then I added the functionality to read and then export the parameter data to a CSV file so I can do parameter audits on the existing drives. To my knowledge, a Rockwell product doesn't that has this functionality to more than one drive at a time. Good luck coding and I am watching this library with interest

closed 2021-09-03 by JL00001

ottowayi, 2021-09-03:
That is because you're setting the `disable` flag on only your logger, all the pycomm3 loggers begin with `pycomm3`.  You can get the library's root logger by doing `logging.getLogger('pycomm3')`.  I think what would want is something like:
```python

pycomm3_logger = logging.getLogger('pycomm3')
pycomm3_logger.propagate = False

...

pycomm3_logger.propagate = True

```

I haven't ever had to temporarily disable logging before, so I'm not sure what the best method is.  I think using the `logging.disable()` function is what you want if you want to disable all logging across the board.  And if you want to disable just the pycomm3 one, the `.propagate = False` property is the way to go.  The `disabled` property doesn't seem to be documented, so without looking at the source I don't know exactly how it works.

ottowayi, 2021-09-03:
@JL00001 You beat me by a few seconds lol glad you got it working.  That drive script sounds pretty cool, I did something similar with backing up parameters for a bunch of drives.  I have plans of some things I want to do, but getting the time is the hard part.  One of those things includes a better CIP object library so you can just call whatever service you want on an object instead of having to build a `generic_message` request.  Hopefully it'll make messaging non-PLC's like drivers easier.

JL00001, 2021-09-03:
I will make the recommended changes you mentioned above. Is there a chance of this library being or could be multi-thread able? I wanted to try multi-treaded programming and being able to have a read/write task to thread could speed up large network task. I tried to do it before; but I didn't know what I was doing

ottowayi, 2021-09-03:
The library itself is not multi-threaded, but it can be used with multi-threading.  Each driver is only 1 connection due to some protocol specifics with sequence numbers and connection identifiers.  You can create multiple drivers that connect to the same PLC though.  What I've done in the past is make multiple threads, each with one instance of a driver.  Then used a `Queue` to add 'jobs' for the drivers, like reading or writing sets of tags. Each thread then waits for jobs to be added to the queue.  You'll have to roll your own implementation though, it's not part of the library.  I would like to eventually add an example people can reference, but don't think I'll add it to the library itself.  It would be hard to maintain and what I want might not be what others want.  If you do implement it, the tip on [this page](https://docs.pycomm3.dev/en/latest/api_reference/logix_driver.html) may be helpful.