ottowayi/pycomm3 #173: [BUG] - CIP Security device / PLC has unusual List Identity Object

MikeDausch, 2021-10-15:
**Pre-checks**
 running latest version 1.2.2

**Description**
PLC  with CIP Security does not include or has unusual  List Identity Object.
1756-L85E v32 and v33
Using front port only.
Path through 1756-EN2T to backplane to card has no issue.

Flashed down to v31 no issues.

**Target PLC**
Model:  1756-L85E
Firmware Revision: v32 and v33

**Code Sample**
def connect_plc():
    with LogixDriver('192.168.1.15/1') as plc:

This works:
def connect_plc():
    with LogixDriver('192.168.1.15/1', init_tags=False, micro800=False, init_info=False) as plc:

but i have to call the .get_tag_list()
and there is no "name" object

ottowayi, 2021-10-16:
Unfortunately, CIP Security is not supported because it is not part of the open protocol specification and I haven't been able to find a copy of the spec online.  I maintain this library as a hobby in my free time, so purchasing the spec and additional hardware is a big cost for something I give away for free.  If I am able to get a copy of the spec in the future, I will attempt to add support.

You may be able to work around the initialization issues by subclassing the driver like:

```python

class MyLogixDriver(LogixDriver):

    def _initialize_driver(self, init_tags, init_program_tags):
        self.__log.info("Initializing driver...")

        if init_tags:
            self.get_tag_list(program="*" if init_program_tags else None)

        self.__log.info("Initialization complete.")

```

I removed the pieces (identity and program name) that you said were failing, but that also assumes all the other services still work correctly too or you might just run into more problems.  Sorry I don't have a better answer for you at this time.  If you can help me by confirming what features do and don't work (with verbose log captures would be great), I could add a flag or something to disable the features that don't work (like how I do for Micro800s).  Long term I would like to support for CIP Security, but at this time I don't have any way of doing it.

MikeDausch, 2021-10-18:
I really like what you have created here. I understand your position. If you can change this to an "enhancement" please do.  I'll do some more research on this.  Browsing through a comm card still works so there is a work around.

MikeDausch, 2021-11-12:
Apparently I have been using an old version the whole time. 0.7.1  I apologize for wasting your time on this.
All paths and processors work just fine with 1.2.4

I was updating using the 
pip install --upgrade pycomm3 command with no issues.  
I needed to do one extra step inside PyCharm.  Click the Upgrade button on the Python Interpreter!

[image]

closed 2021-11-12 by MikeDausch

ottowayi, 2021-11-12:
Ohh that's great actually lol glad it was that simple