ottowayi/pycomm3 #186: [BUG] - Predefined tags

SISAutomationIMA, 2021-11-12:
Hello,

I find that in **logix_driver.py** in the **_parse_template_data** function, there is no predefined tag.
```
predefine = template_name is None
if predefine:  # predefined types put name as first member (DWORD)
    template_name = member_names.pop(0)
```
The boolean is never set to true.
The problem is that it is in this case impossible to read some predefined tags.
For example in a **TIMER** type tag, the **control** tag makes reading impossible. In the **internal_tags**, the tags **EN**, **TT**, **DN** have their offset inside the tag DINT **Control**.
```
'name': 'TIMER',
'internal_tags': {
    'Control': {'offset': 0, 'tag_type': 'atomic', 'data_type': 'DINT', 'data_type_name': 'DINT', 'array': 0, 'type_class': DINT},
    'PRE': {'offset': 4, 'tag_type': 'atomic', 'data_type': 'DINT', 'data_type_name': 'DINT', 'array': 0, 'type_class': DINT},
    'ACC': {'offset': 8, 'tag_type': 'atomic', 'data_type': 'DINT', 'data_type_name': 'DINT', 'array': 0, 'type_class': DINT},
    'EN': {'offset': 3, 'tag_type': 'atomic', 'data_type': 'BOOL', 'data_type_name': 'BOOL', 'bit': 7, 'type_class': BOOL},
    'TT': {'offset': 3, 'tag_type': 'atomic', 'data_type': 'BOOL', 'data_type_name': 'BOOL', 'bit': 6, 'type_class': BOOL},
    'DN': {'offset': 3, 'tag_type': 'atomic', 'data_type': 'BOOL', 'data_type_name': 'BOOL', 'bit': 5, 'type_class': BOOL}
}
```
When I try to read a Timer Tag, I have the error **Failed to parse reply -**.

Do you know how to fix this ?

Thank you

ottowayi, 2021-11-12:
What version is your PLC?  Because I was able to successfully read a TIMER in my test PLC.  For the predefined types the bool host was usually called `CTL` and not `Control` like in your example, so that makes me think something has changed in maybe a newer firmware?  The spec from Rockwell explicitly says that predefine types should not be accessed as a whole because they have more complex rules than UDTs.  So any support for them is hacks trying to make it work and a change in the firmware could break it without warning.

ottowayi, 2021-11-12:
But... this did lead my to find a bug in the boolean member offsets for the predefined types.  I have some timers and other predefined tags in my test PLC program, but for some reason I neglected to include them in the online unit tests.  I will be fixing that right away.

ottowayi, 2021-11-12:
hmm I think this means I'll need to add some other way of determining if a tag is a predefined type, I would like to avoid having to maintain a list of hardcoded predefined types.  Unfortunately I'm not longer in controls so I don't have easy access to a variety of hardware anymore.  I only have a really on CompactLogix, but I'm going to see if there is something I can track down to try and recreate this.

SISAutomationIMA, 2021-11-12:
I'm trying to code a way to add a Boolean in **_bit_members** instead of **_struct_members** if their offset can be inside another **_struct_members**.

ottowayi, 2021-11-12:
I think I may have found a flag bit in the symbol type for predefined tags, I just pushed the change to develop if you want to try it out.

SISAutomationIMA, 2021-11-15:
Hi,

I tried with your update but it's not working.
The fault comes from this section :
```
780 |    # range of non-predefined structs is 0x100 - 0xEFF according to spec
781 |    # so if outside that range assume it is a predefined type
782 |    predefine = _type < 0x100 or _type > 0xEFF
783 |    if predefine:  # predefined types put name as first member (DWORD)
784 |    
785 |        template_name = member_names.pop(0)
```
The first member is not the **template_name** so the predefined data type is wrong.
I test it with the data type **TIMER** and **STRING** and I wasn't able to read it. 
So I just commented these lines and it works well.
But I'm afraid that modification won't worked with lower firmware than 32.00.

What do you think ?

ottowayi, 2021-11-15:
ohh I wonder if the newer versions of these store the template name just like normal UDTs do, meaning `template_name` already has a value.. what about changing ln 783 to `if template_name is None and predefine:`?

ottowayi, 2021-11-15:
Awesome thanks, I'll verify it still works on lower versions and get a fix out.

SISAutomationIMA, 2021-11-15:
ok thank you

ottowayi, 2021-11-16:
fixed in v1.2.5

closed 2021-11-16 by ottowayi

ottowayi, 2021-12-01:
@SISAutomationIMA I have fixed another bug related to how bools are mapped inside structures.  It simplifies the code a lot by removing the tracking of host members and operates directly on the attribute offsets.  Would you be willing to test the `develop` branch against this issue?  I'd like to confirm it works on v32+ before releasing it.

SISAutomationIMA, 2021-12-06:
Alright I'll watch it all.
I will keep you posted.

SISAutomationIMA, 2021-12-06:
I test it with a Controller Logix 32.13 and It works perfectly.
I look the code I think it is now more simple and more understandable.
This is a great update.

ottowayi, 2021-12-06:
great, thank you!  I'm still working on finding a more modern processor to test against, so I appreciate you being able to do that for me.  I made some poor design decisions early on that lead to some really complicated code, I think this is much simpler and maintainable.  I'll work on getting it cleaned up and released.